OpenTracksApp/OpenTracks #596: Doesn't run in background

mbnoimi, 2021-02-10:
**Describe the bug**
After upgrading my phone to Android 10. OR no longer able to run in background (while other apps work fine ex. OsmAnd)

**My Android Configs**
[image]

[image]

[image]
OsmAnd works fine. Both OsmAnd & OT have the same configs while the only difference OsmAnd has **Allow all the time** location permission.

**Technical information**
 - Device: Google Pixel 5
 - OS: Android 10 (Q)
 - Version 3.14.3

mbnoimi, 2021-02-10:
BTW, I tried to modify **Battery optimization**, but it didn't affect on the result (as soon as I turn off my screen OT no longer run in background)
[image]
[image]

dennisguse, 2021-02-10:
We had some similar reports, but there is no real fix as there is not too much what we can do.
On my LineageOS phone, the battery saver, for example, prevents usage of GPS.
And there is no error message :shrug: 

I just know that the location permissions should be okay (OSMAnd is not too privacy friendly here).
Some more details can be found here: https://github.com/OpenTracksApp/OpenTracks/issues/282#issuecomment-646692317

dennisguse, 2021-02-13:
Please reopen, if we can be of more assistance.

closed 2021-02-13 by dennisguse