OpenTracksApp/OpenTracks #597: GPX: enhance export/import

dennisguse, 2021-02-10:
After the refactoring of #316, only KML/KMZ exports all information to export and restore a track.
The implementation for GPX was reverted as for every trkpt GPX requires the presence of lat/lon (in difference to KML/KMZ).

Some more info can be found here: #587

Goal: be able to export and import a track while restoring all details.

vanous, 2021-07-11:
Let me use this issue for a quick question @dennisguse please. Is there a way to export a single/multiple GPX from OT? 

Background: we allow exporting GPX from [Gadgetbridge](https://codeberg.org/Freeyourgadget/Gadgetbridge/), but importing and using it was not really possible... but now it will be [easier/possible](https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/2341) to import and link a GPX to a recorded activity.

Thank you 😊

dennisguse, 2021-07-11:
@vanous Yes, but only manually via the settings.
Or using the automatic export after a finishing a recording.

vanous, 2021-07-11:
This is interesting, do you mean the `Settings → Tracks → Export all`?
The automatic export is also very interesting. I will explore it and document to our wiki :)
Thank you very much!

vanous, 2021-07-18:
Hi @dennisguse 

﻿based on some current work to make smoother integration between Gadgetbridge and other apps, i wrote a manual on how to integrate better with FitoTrack and OpenTracks.

Here is a blog post that describes it in a nutshell https://blog.freeyourgadget.org/add_gps_track_to_any_sports_activity.html

And here is the manual on how to do this: https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Integrating-with-Sports-Tracking-apps

Please feel free to check it and let me know if something is missing or not correct.

:)

dennisguse, 2021-07-18:
@vanous Looks great!

Only two suggestion: 
* wiki: splitting the page into sub-pages per application might make it easier to consume.
The structure is not that easy to understand as the headlines are not numbered
* blog: adding a caption to the images might help.
I was for example not sure, if the last screenshots are from Gadgetbridge or rather OSMAnd...

And another nitpick: it is usually better to not hijack issues; just open a new one or open a discussion thread :)

vanous, 2021-07-18:
Added the labels, but have to think about the splitting, yes, the page is much too long but md is not good for this, `imports` are missing... And very sorry about not realizing there is the `Discussions` now in GH, my bad. Feel free to delete this from here, to clean it up, next time i will remember to use Discussions.

rgmf, 2021-07-20:
@vanous fantastic work, and very useful 👏 👏 👏

closed 2022-01-09 by dennisguse