OpenTracksApp/OpenTracks #601: Normalize sensors stats (avg)

rgmf, 2021-02-14:
**Is your feature request related to a problem? Please describe.**
Following with https://github.com/OpenTracksApp/OpenTracks/pull/592 the sensors stats averages has to be normalized.

**Describe the solution you'd like**
The solutions are in the follow comments:
https://github.com/OpenTracksApp/OpenTracks/pull/592#issuecomment-775488089
https://github.com/OpenTracksApp/OpenTracks/pull/592#issuecomment-778795768

rgmf, 2021-02-18:
I take notes about the progress of this issue:

I've been playing with sqlite and the possibility of normalize the avg. This SQL sentence is the nearly I have but it's difficult because the several types of the trackpoints.

I don't know if it worths or it's better doing the normalization in java code instead of sql.

```sql
select avg(x.sensor_heartrate * (x.time - coalesce(y.time, x.time))) / 1000 from (
       select w.time, max(d.time) previous, w.sensor_heartrate
       from trackpoints w left join trackpoints d on d.time < w.time
       where w.type=0
       group by w.time) x
left join trackpoints y on y.time=previous
where y.type=0
order by x.time
```

Basically what this SQL sentence does is ponderate the heart rate based on the time this heart rate last.

dennisguse, 2021-02-18:
(about code vs SQL)
Don't know; both work. Back in 2012 I would have said the DB might be faster, but nowadays :man_shrugging: 

Tiny fixe to the SQL:
* ORDER BY  x._id 

Now comes the ugly part.
In theory, all trackpoints may have sensor data (even SEGMENT_MANUAL_START, e.g., PAUSE/RESUME case). 
I guess the only exception is the very first SEGMENT_MANUAL_START as the sensor connection is established afterwards.
So, we have the following cases:
* first trackpoint of a track (ignore likely has no data)
* first tackpoint in segment (SEGMENT_AUTOMATIC_START) time difference to previous trackpoint as it is just for segment splitting (i.e., rendering and distance computation if not present via sensor).
* all other trackpoints: use.

And another ugly one: what happens if not _all_ trackpoints have sensor data (e.g., sensor got disconnected; i.e., sensor data null).

(side note)
SQL (for me) is easier to read if SQL keywords are upcase.
Then it also works without code highlighting (for example as String in Java) ;)

dennisguse, 2021-02-18:
Forgot to mention it: for the indoor cycling (#500), we might end up with a  SEGMENT_MANUAL_END that contains sensor data.
Initial implementation would be:
* start: SEGMENT_MANUAL_START
* end: SEGMENT_MANUAL_END (with most recent sensor data for HR and sum(distance))

When adding non-GPS (normal) trackpoints there will be some in between, but I guess the SEGMENT_MANUAL_END is likely to still contain data (although it might not; like sensor disconnect).

rgmf, 2021-02-19:
@dennisguse the query I posted take about 50 seconds with a track that hast about 1500 trackpoints... it's too slow.

I'm going to propose a coding option. In that way is simpler to handle all the possibilities. Anyway, my SQL skills are not too much good and I don't know if we can build an efficient SQL query that handle all the possibilities.

dennisguse, 2021-02-19:
@rgmf wow that is really slow.
I mean everything above 250ms would be unacceptable from a user perspective anyhow.
I guess it is problematic for the database that there is no index on `time` and also the two joins.

Here is an untested version using the `_id` (physical storage layout; should be fast) and one join
```
SELECT AVG(x.sensor_heartrate * (t.time - COALESCE(t1.time, t.time))) / 1000
      (SELECT _id, time FROM trackpoints t1
       WHERE t1._id < t._id
       ORDER BY _id LIMIT 1) lag_time,
FROM trackpoints t
WHERE t.trackId = 1
AND t.type NOT IN (-2)
```

But we can also move to Java-based computation.
Way easier to understand.
SQL with a lot of joins becomes write only...

PS: Never used COALESCE before!
Thanks!

rgmf, 2021-02-19:
> Here is an untested version using the _id (physical storage layout; should be fast) and one join

Awesome!!! It takes 27ms for a track with more than 7000 trackpoints!!

```sql
SELECT avg(t.sensor_heartrate * (
           (SELECT t1.time
            FROM trackpoints t1
            WHERE t1._id > t._id
            ORDER BY _id LIMIT 1) - t.time)) / 1000
FROM trackpoints t
WHERE t.trackId = 4
AND t.type NOT IN (-2)
```

> PS: Never used COALESCE before!
> Thanks!

Welcome. Whatever happens always learning something, at least :smile:

dennisguse, 2021-02-19:
Awesome!

I forgot one WHERE: the sub-select should also filter by trackId as otherwise we might get a trackpoint from a previous track. I guess then the COALESCE is needed again.

rgmf, 2021-02-19:
> Awesome!
> 
> I forgot one WHERE: the sub-select should also filter by trackId as otherwise we might get a trackpoint from a previous track. I guess then the COALESCE is needed again.

Yes, I forgot the COALESCE... I've updated the query in the post.

closed 2021-03-03 by rgmf