OpenTracksApp/OpenTracks #602: Support for ANT+ and Classic Bluetooth

7eter, 2021-02-17:
Why was support for ANT+ and Classic Bluetooth removed with the rework of _My Tracks_?
I would love to use my old _Zephyr_ heart rate sensor with _OpenTracks_.
Is there any workaround to get the 'old' sensors to work?

dennisguse, 2021-02-17:
TLDR: there is no easy way and this only for Bluetooth classic.
If you really want this it, feel free to invest the time.
You could try using the old stack from MyTracks and bringing together with the new Bluetooth LE stack.
But I expect it to be rather painful experience (with a lot of learnings).

Actual explanation:
I decided that initially, when I started reworking MyTracks.
I basically removed everything that was not really useful or would be hard to maintain.
Some things survived and we struggle nowadays with it (like internal database access).

ANT+ was removed as it requires a proprietary library.

For Bluetooth classic, the answer is slightly more  complicated.
Nowadays the standard is Bluetooth LE and this replaced Bluetooth classic more or completely.
In addition, the connection setup and handling for Bluetooth classic is way different than Bluetooth LE.
So, implementing and maintaining a joint Bluetooth stack for both variants would be kind of complicated. It starts with simple things like sensor selection in the settings.

Does this help?

7eter, 2021-02-19:
Sounds like a frustrating job - and I can see why the motivation is not to high given that all new devices use BLE. 
I'll guess I will live with the old MyTracks and import the recorded tracks to OpenTracks - as this works quite well. 
Thanks for the detailed explanation!

closed 2021-02-19 by 7eter

dennisguse, 2021-02-19:
i was actually curious how the code looked like.
And for Bluetooth Classic every company was implementing their own protocol.
So, MyTrack only supported Zephyr and Polar.
Nowadays I am really happy with the Bluetooth LE solution - for HR it is really easy :)
Now I understood why ANT+ was created considering BLE was not yet there.

@7eter Using MyTracks is actually a really good workaround!

You can also upgrade to an early OpenTracks version before this commit the removal of the Bluetooth Classic stack: https://github.com/OpenTracksApp/OpenTracks/commit/3a685c609d877f0c7bba52c46b1dedb75ca2261e
There was a lot of analytics code and I fixed some other stuff.

7eter, 2021-02-19:
oh cool I'll check that out - thank you again! :)