OpenTracksApp/OpenTracks #607: app not tracking activity

andrearicci, 2021-02-22:
I use it when I go to work by bike. It takes me 50 minutes, but the app records only the first 200 meters.

**To Reproduce**
1. turn on gps
2. start tracking
3. after arrival, stop tracking
4. recorded 50 minutes, 5 minutes in motion, track recorded from house to 200 m about

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: Xaiaomi redmi note 4
 - OS: miui
 - Version 11.0.2

rgmf, 2021-02-22:
Other users have been trouble with battery saving. I don't know if it is related with your issue. If so, this can help you: https://github.com/OpenTracksApp/OpenTracks/issues/177

andrearicci, 2021-02-23:
ok - that fixed it

closed 2021-02-23 by andrearicci