OpenTracksApp/OpenTracks #613: App dies when hitting record

writingpoli, 2021-02-27:
**Describe the bug**
App dies when trying to start tracking walk.

**To Reproduce**
1. Open app
2. Click on record
3. See error

**Technical information**
 - LG thinQ

[Android]
Time Now = Fri Feb 26 19:29:10 CST 2021
Android ID = a7bc27fbb4ef7305
UUID Random = 27687910-4900-4587-9173-510a32582230
Google™ Play Store = Version 24.1.31-21 [0] [PR] 357815498
Google™ Play Services = Version 21.02.14 (120400-352619232)
EMULATOR_DEVICE = false
FINGERPRINT = lge/alphalm_lao_com/alphalm:10/QKQ1.190929.002/8150100201201:user/release-keys
MANUFACTURER = LGE
BRAND = lge
MODEL = LM-G820
PRODUCT = alphalm_lao_com
DISPLAY = QKQ1.190929.002
SERIAL = unknown
DEVICE = alphalm
BUILD ID = QKQ1.190929.002
BUILD_TIME = 23 Nov 2020 09:25 PM CST
TAGS = release-keys
VERSION.CODENAME = REL
VERSION.INCREMENTAL = 8150100201201
VERSION.RELEASE = 10
VERSION.BASE_OS = lge/alphalm_lao_com/alphalm:10/QKQ1.190929.002/2024017598fe8:user/release-keys
VERSION.PREVIEW_SDK_INT = 0
VERSION.SECURITY_PATCH = 2020-12-01
SECURITY_PATCH = 2020-12-01
VERSION.SDK_INT = 29
BOOTLOADER = unknown
HARDWARE = alphalm
BOARD = msmnile
RADIO = MPSS.HE.1.0.c4-00090-SM8150_GEN_PACK-1.300496.1.307489.1
RADIO_INTERFACE = Qualcomm RIL 1.0
OPENGL_VERSION = 3.2
HOST = LGEACI5R10
TYPE = user
USER = jenkins
SUPPORTED_ABIS = [arm64-v8a, armeabi-v7a, armeabi]
SUPPORTED_32_BIT_ABIS = [armeabi-v7a, armeabi]
SUPPORTED_64_BIT_ABIS = [arm64-v8a]
PARENTAL_CONTROL_ENABLED = null
PARENTAL_CONTROL_LAST_UPDATE = null
PARENTAL_CONTROL_REDIRECT_URL = null
ACCESSIBILITY_ENABLED = 1
ENABLED_ACCESSIBILITY_SERVICES = com.airmore/com.apowersoft.airmore.service.NotifyAccessibilityService:app.medicalid.free/app.medicalid.lockscreen.AccessibilityService:com.google.audio.hearing.visualization.accessibility.scribe/.SpeechToTextAccessibilityService:com.securityandprivacy.android.verizon.vms/com.mcafee.monitor.MMSAccessibilityService:com.android.settingsaccessibility/.flashalerts.FlashAlertsService:eu.thedarken.sdm/.accessibility.core.ACCService:com.arlosoft.macrodroid/.action.services.UIInteractionAccessibilityService:com.arlosoft.macrodroid/.triggers.services.FingerprintAccessibilityService:nextapp.sp/.monitor.AccessibilityMonitorService:com.arlosoft.macrodroid/.triggers.services.MacroDroidAccessibilityServiceJellyBean:com.microsoft.appmanager/com.microsoft.mmx.screenmirroringsrc.accessibility.ScreenMirroringAccessibilityService:com.a0soft.gphone.acc.free/com.a0soft.gphone.acc.acs.AutoAccessSrvc:com.google.android.accessibility.soundamplifier/.SoundAmplifierService:com.sophos.smsec/.plugin.webfiltering.service.WebAccessibilityService
ACCESSIBILITY_DISPLAY_INVERSION_ENABLED = null
DEFAULT_INPUT_METHOD = com.lge.ime/.LgeImeImpl
SELECTED_INPUT_METHOD_SUBTYPE = 500
ENABLED_INPUT_METHODS = com.lge.ime/.LgeImeImpl;500:com.touchtype.swiftkey.beta/com.touchtype.KeyboardService:com.google.android.googlequicksearchbox/com.google.android.voicesearch.ime.VoiceInputMethodService:
INPUT_METHOD_SELECTOR_VISIBILITY = null
TTS_DEFAULT_PITCH = null
TTS_DEFAULT_RATE = null
TTS_DEFAULT_SYNTH = com.google.android.tts
TTS_ENABLED_PLUGINS = null
ALLOWED_GEOLOCATION_ORIGINS = http://www.google.com http://www.google.co.uk
LOCATION_MODE = 3
SETTINGS_CLASSNAME = null
SKIP_FIRST_USE_HINTS = null
TOUCH_EXPLORATION_ENABLED = 0
SOUND_EFFECTS_ENABLED = No

Global Setting
DEVELOPMENT_SETTINGS_ENABLED = No
Device Encryption = Active But Using Default Key
ADB_ENABLED = No
AUTO_TIME = Yes
AUTO_TIME_ZONE = Yes
DEFAULT_ALARM_ALERT_URI = content://settings/system/ringtone
ALARM_ALERT = null
AIRPLANE_MODE_ON = No
AIRPLANE_MODE_RADIOS = cell,bluetooth,wifi,nfc
ALWAYS_FINISH_ACTIVITIES = No
ANIMATOR_DURATION_SCALE = 1.0
TRANSITION_ANIMATION_SCALE = 1.0
WINDOW_ANIMATION_SCALE = 1.0
BLUETOOTH_ON = Yes
BOOT_COUNT = 98
DATA_ROAMING = Yes
DEVICE_NAME = LGE LM-G820
DEVICE_PROVISIONED = Yes
HTTP_PROXY = null
MODE_RINGER = 2
NETWORK_PREFERENCE = 1
STAY_ON_WHILE_PLUGGED_IN = No
USB_MASS_STORAGE_ENABLED = Yes
USE_GOOGLE_MAIL = null
WAIT_FOR_DEBUGGER = No
WIFI_ON = Yes
WIFI_WATCHDOG_ON = Yes
WIFI_DEVICE_OWNER_CONFIGS_LOCKDOWN = null
WIFI_MAX_DHCP_RETRY_COUNT = 3
WIFI_MOBILE_DATA_TRANSITION_WAKELOCK_TIMEOUT_MS = 0
WIFI_NETWORKS_AVAILABLE_REPEAT_DELAY = 0
WIFI_NUM_OPEN_NETWORKS_KEPT = 0
WIFI_SLEEP_POLICY = 2
LOCALE_LANGUAGE = English (United States)
LOCALE_COUNTRY = United States
LOCALE_ID = America/Chicago
LOCALE_TIMEZONE = Central Standard Time
LOCALE_DAYLIGHT_SAVING = 3600000
LOCALE_USE_DST = true
LOCALE_RAWOFFSET = -6.0 hrs
LOCALE_CURRENCY = USD

[Top]

dennisguse, 2021-02-27:
Which version? And can you provide a logcat?
The info above (sadly) does not help for finding the issue.

oceyral, 2021-02-27:
Hi @dennisguse ,

Same thing here on app version 3.15.0, android 10 : app crash when hitting record. Empty track segments are created.

Not quite comfortable about sharing logcat, as I don't know what informations are in there, nor how to filter it, however looking at it briefly, the culprit seems obvious : 

```
FATAL EXCEPTION: main
Process: de.dennisguse.opentracks, PID: 24582
java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
	at de.dennisguse.opentracks.content.data.TrackPoint.getLatitude(TrackPoint.java:226)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment.setLocationValues(StatisticsRecordingFragment.java:446)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment.updateUI(StatisticsRecordingFragment.java:259)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment.lambda$onTrackUpdated$3$StatisticsRecordingFragment(StatisticsRecordingFragment.java:164)
	at de.dennisguse.opentracks.fragments.-$$Lambda$StatisticsRecordingFragment$qTCJqbKQ9ik7N-JSKTl9FPND1dg.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7615)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)
```

This is quite bad, as the app can't even be rolled back to a previous version because of the database migration.

dennisguse, 2021-02-27:
@oceyral Thanks!
That was sufficient!

The problem occurs with v3.15.0 if you Settings->Show Latitude/Longitude activated.
If you deactivate, the crash does not occur.

I will create a new release today!
The F-Droid release might take 2 weeks.

oceyral, 2021-02-27:
Great ! Thank you for all the time and effort you dedicate to this project !

closed 2021-02-27 by dennisguse

dennisguse, 2021-02-27:
Issues is fixed and released in v3.15.1
F-Droid availability will take some days.

PS: there was actually another bug: lat/lon where actually never shown as the UI considered the data old.
This is fixed as well.

dennisguse, 2021-03-04:
v3.15.1 is live on F-Droid since 03-Mar-2021.

commit 4040e3f4 (2021-02-27): Bugfix: crash on recording start due to show latitude/longitude option.

Fixes #613.