OpenTracksApp/OpenTracks #614: Bug: GPX export crashes on multi-track export

dennisguse, 2021-02-27:
Reported via GooglePlay Store.

Somehow one can trigger a multi-track export to GPX.
And that is not supported by the GPXTrackExporter.

```
java.lang.RuntimeException: 
  at android.os.AsyncTask$4.done (AsyncTask.java:399)
  at java.util.concurrent.FutureTask.finishCompletion (FutureTask.java:383)
  at java.util.concurrent.FutureTask.setException (FutureTask.java:252)
  at java.util.concurrent.FutureTask.run (FutureTask.java:271)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:919)
Caused by: java.lang.UnsupportedOperationException: 
  at de.dennisguse.opentracks.io.file.exporter.GPXTrackExporter.writeTrack (GPXTrackExporter.java:108)
  at de.dennisguse.opentracks.content.provider.ShareContentProvider.lambda$openFile$1 (ShareContentProvider.java:228)
  at de.dennisguse.opentracks.content.provider.-$$Lambda$ShareContentProvider$UUYLBGQZWBfxU6qi91pkc6uXicw.writeDataToPipe (Unknown Source:11)
  at android.content.ContentProvider$1.doInBackground (ContentProvider.java:2033)
  at android.os.AsyncTask$3.call (AsyncTask.java:378)
  at java.util.concurrent.FutureTask.run (FutureTask.java:266)
```

dennisguse, 2021-02-28:
I refactored GPXTrackExporter as it cannot export multiple tracks.
But the ShareContentProvider always tries to export multiple tracks (which is not supported for GPX) leading to this crash.

I guess, #615  is very related.

rgmf, 2021-02-28:
> I guess, #615 is very related.

I think so. Maybe this one will fix #615

dennisguse, 2021-02-28:
Forget what I said: GPX __can__ export multiple tracks.
We just never implemented this.
See #617.

rgmf, 2021-02-28:
In Android 10 if you try to share a track you get this error but track is shared correctly:

```
2021-02-28 18:03:08.566 5735-5752/de.dennisguse.opentracks.debug E/DatabaseUtils: Writing exception to parcel
    java.lang.SecurityException: Permission Denial: reading de.dennisguse.opentracks.content.provider.ShareContentProvider uri content://de.dennisguse.opentracks.debug.content/tracks/gpx/338/Garganta%2C%20Boquera%2C%20Bay%C3%B3n%2C%20Albatera.gpx from pid=5557, uid=1000 requires the provider be exported, or grantUriPermission()
        at android.content.ContentProvider.enforceReadPermissionInner(ContentProvider.java:729)
        at android.content.ContentProvider$Transport.enforceReadPermission(ContentProvider.java:602)
        at android.content.ContentProvider$Transport.query(ContentProvider.java:231)
        at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:104)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
```

closed 2021-02-28 by dennisguse

dennisguse, 2021-03-01:
Released as v3.15.2; will be live on Google today; F-Droid: at least two days.

dennisguse, 2021-03-06:
v3.15.2 is live on F-Droid.

PR #616 (2021-02-28): GPX export: re-implemented basic support for multiple tracks.
Fixes #614.
commit 8a358357 (2021-02-28): GPX export: re-implemented basic support for multiple tracks.

Fixes #614.