OpenTracksApp/OpenTracks #615: Shared gpx-Files are empty

MartinHH, 2021-02-28:
**Describtion**
When using the share button of a track, the file that is transferred to the target app is empty (but has the correct file name). This did not occur with 3.14.3, but now occurs with 3.15.0 (irregardless of whether the track was recorded before or after updating to 3.15.0).

**To Reproduce**
1. From the main view, open a single (recorded) track
2. Click on the share icon.
3. Select an app to share to (e.g. mail)
4. In the target app, the file (e.g. the mail attachment) has the name of the track (with suffix `.gpx`), but it is empty (0 kb size)

I'm not an android expert, but this stacktrace from logcat looks somewhat related:
```
2021-02-28 12:53:54.228 15524-15524/? W/Bundle: Key android.intent.extra.TEXT expected ArrayList<CharSequence> but value was a java.lang.String.  The default value <null> was returned.
2021-02-28 12:53:54.229 15524-15524/? W/Bundle: Attempt to cast generated internal exception:
    java.lang.ClassCastException: java.lang.String cannot be cast to java.util.ArrayList
        at android.os.BaseBundle.getCharSequenceArrayList(BaseBundle.java:1237)
        at android.os.Bundle.getCharSequenceArrayList(Bundle.java:1024)
        at android.content.Intent.getCharSequenceArrayListExtra(Intent.java:7323)
        at android.content.Intent.migrateExtraStreamToClipData(Intent.java:10168)
        at android.content.Intent.migrateExtraStreamToClipData(Intent.java:10107)
        at android.app.Instrumentation.execStartActivity(Instrumentation.java:1621)
        at android.app.Activity.startActivityForResult(Activity.java:4762)
        at androidx.fragment.app.FragmentActivity.startActivityForResult(FragmentActivity.java:675)
        at android.app.Activity.startActivityForResult(Activity.java:4702)
        at androidx.fragment.app.FragmentActivity.startActivityForResult(FragmentActivity.java:662)
        at android.app.Activity.startActivity(Activity.java:5123)
        at android.app.Activity.startActivity(Activity.java:5091)
        at de.dennisguse.opentracks.TrackRecordedActivity.onOptionsItemSelected(TrackRecordedActivity.java:151)
        at android.app.Activity.onMenuItemSelected(Activity.java:3707)
        at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:383)
        at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:228)
        at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
        at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
        at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:65)
        at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:207)
        at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:779)
        at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:834)
        at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
        at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:985)
        at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:975)
        at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:623)
        at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:151)
        at android.view.View.performClick(View.java:6291)
        at android.view.View$PerformClick.run(View.java:24931)
        at android.os.Handler.handleCallback(Handler.java:808)
        at android.os.Handler.dispatchMessage(Handler.java:101)
        at android.os.Looper.loop(Looper.java:166)
        at android.app.ActivityThread.main(ActivityThread.java:7529)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:245)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:921)
```

**Technical information**
 - Device: HUAWEI P10 Lite (WAS-LX1A)
 - OS: EMUI 8.0.0 / Android 8.0.0 (latest available Huaweii stock OS for the device)
 - Version 3.15.0

dennisguse, 2021-02-28:
You wanted to share _one_ track right?

rgmf, 2021-02-28:
I think the error is this. I get the same logcat information that @MartinHH but also this one:

```
2021-02-28 17:21:48.506 12820-12837/de.dennisguse.opentracks.debug E/DatabaseUtils: Writing exception to parcel
    java.lang.SecurityException: Permission Denial: reading de.dennisguse.opentracks.content.provider.ShareContentProvider uri content://de.dennisguse.opentracks.debug.content/tracks/kmz_with_trackdetail_and_sensordata_and_pictures/1/2021-02-28%2017%3A04.kmz from pid=10771, uid=1000 requires the provider be exported, or grantUriPermission()
        at android.content.ContentProvider.enforceReadPermissionInner(ContentProvider.java:729)
        at android.content.ContentProvider$Transport.enforceReadPermission(ContentProvider.java:602)
        at android.content.ContentProvider$Transport.query(ContentProvider.java:231)
        at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:104)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
```

rgmf, 2021-02-28:
@dennisguse I'm trying to figure out how to fix it...

dennisguse, 2021-02-28:
We also have this one: https://github.com/OpenTracksApp/OpenTracks/issues/614

MartinHH, 2021-02-28:
> You wanted to share _one_ track right?

Yes, just one track (and I would not know how to share more than one track at a time).

I had another look at the logcat and did not see `SecurityException` that @rgmf posted. However I did find this:

```
2021-02-28 17:41:14.843 25110-25349/? E/AndroidRuntime: FATAL EXCEPTION: AsyncTask #1
    Process: de.dennisguse.opentracks, PID: 25110
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$3.done(AsyncTask.java:365)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
        at java.util.concurrent.FutureTask.run(FutureTask.java:271)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:784)
     Caused by: java.lang.UnsupportedOperationException
        at de.dennisguse.opentracks.io.file.exporter.GPXTrackExporter.writeTrack(GPXTrackExporter.java:108)
        at de.dennisguse.opentracks.content.provider.ShareContentProvider.lambda$openFile$1(ShareContentProvider.java:228)
        at de.dennisguse.opentracks.content.provider.-$$Lambda$ShareContentProvider$UUYLBGQZWBfxU6qi91pkc6uXicw.writeDataToPipe(Unknown Source:11)
        at android.content.ContentProvider$1.doInBackground(ContentProvider.java:1844)
        at android.os.AsyncTask$2.call(AsyncTask.java:345)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636) 
        at java.lang.Thread.run(Thread.java:784) 
2021-02-28 17:41:14.844 648-648/? I/GRALLOC: 	 shrFd=10,fmt=0x1,intFmt=0x1,btStrd=4608,size=2433024,pid=648 
```

dennisguse, 2021-02-28:
Thanks!
That is identical to #614.

@MartinHH you can select multiple tracks in the list.
But for GPX that always will lead to a crash until v3.15.1.

dennisguse, 2021-02-28:
Should be fixed with https://github.com/OpenTracksApp/OpenTracks/pull/616
#617 should improve the situation.

@MartinHH Please re-open, if the changes did not solve your issue.
And thanks for providing the logcats; this was very helpful!

closed 2021-02-28 by dennisguse