OpenTracksApp/OpenTracks #619: elevation gain/loss without barometer

hex-m, 2021-03-01:
**Is your feature request related to a problem? Please describe.**
My device shows "elevation" but shows neither "gain" nor "loss". This is confusing to me.

**Describe the solution you'd like**
I understand that devices without barometer (like mine) are not good at measuring elevation/altitude. But if it's good enough to show the current elevation, it should also be good enough for calculating elevation gain and loss. Maybe the accuracy could be shown?

**Describe alternatives you've considered**
Elevation could be hidden altogether on devices without barometer.

dennisguse, 2021-03-01:
This would be a great addition, but it is actually quite complicated to do (if done properly).
Here is some more info: https://www.gpsvisualizer.com/tutorials/elevation_gain.html

So, if you really want to work on that: go ahead!
I can give you some info on how to start.

PS: the issue is that elevation from a GPS is "okay", but not sufficient to directly compute the gain/loss as the noise is too high and adds up.
So, showing the absolute value is useful.

dennisguse, 2021-03-17:
It might not be that complicated - it might be sufficient to just change TrackStatisticsUpdater, so it filters the TrackPoint.getElevation(). A similar approach is already in place for the speed.
And we only would need to this, if no barometric data is present.

dennisguse, 2021-09-24:
Well then let's build this one first (before grade/slope #955).
I can prepare something next month.
My problem is a severe lack of mountains (60m above ground doesn't count :) )and thus test data.
@vsevjednom-cz could you then take care about the testing and fine tuning?

https://github.com/OpenTracksApp/OpenTracks/issues/955#issuecomment-926464206

vsevjednom-cz, 2021-09-24:
> could you then take care about the testing and fine tuning?

Of course 🤩. There aren't many mountains here around Brno, but I know all of existing ones 🚲.

rgmf, 2021-09-24:
@dennisguse @vsevjednom-cz I can help testing it too.

Niekto350, 2022-01-15:
I've bought a cheap phone to use as a bike computer / hiking tracker without the sensor and I have the same issue. I understand that it's not implemented because of the noise of the GPS data.

However I was thinking a bit about it and I think I've found a solution. What about calculating a moving average of the elevation data based on gps and use that for the calculation? Something similar should be already done when generating elevation graph. The data there is not all over the palce but looks quite usable for at least an approximate calculation.

dennisguse, 2022-01-15:
@Niekto350 had no time to work on this so far, but the approach described here sounded very promising: https://github.com/OpenTracksApp/OpenTracks/issues/619#issuecomment-788071078

Niekto350, 2022-01-15:
Thanks for the info.

And if we happen to need this for some reason than here is more info including a formulas for the approach I was talking about https://www.investopedia.com/terms/m/movingaverage.asp including EMA version (which should be more responsive a probably a better fit for this usecase). It's commonly used to smooth out the data for stock trading which are quite similar compared to the gps data we're working with.

dennisguse, 2022-12-08:
I guess, we will go with #1424 to actually get a barometric sensor if the phone has none.
Deriving gain/loss from GPS is way too imprecise.

closed 2022-12-08 by dennisguse