OpenTracksApp/OpenTracks #620: gpx export/sharing fails

Grarie, 2021-03-01:
**Describe the bug**
On the current version (see below) whenever I want to export a track as gpx it fails, i.e. no file is created or if I try to share it in a messenger it's an empty file or nothing happens at all.

Interestingly it's just the gpx export. kml works fine and i complete as far as I can tell.

**To Reproduce**
1. Go to main screen with the track list
2. Click on a track -> track details are shown
3. Click on "share"-symbol in the upper right corner
4. Chose any other app (WhatsApp, Telegram, Amaze, ...)

Expected result: file is saved on device or sent via messenger

Actual result: nothing / error from the other app (file not supported) depending on the app 

I could add logging information if you tell me what to do.

**Technical information**
 - Device: Samsung Galaxie S10e
 - OS: Stock-Android (Samsung flavour) 11
 - Version 3.15.0-418f092db / Version Code 3855

dennisguse, 2021-03-01:
Duplicate of #614.
Changes not yet released.

Only option: share as KML or KMZ.
I broke GPX export.

PS: thanks for the detailed report.

closed 2021-03-01 by dennisguse

Grarie, 2021-03-01:
Thank you for the fast reply. Should have found the other issue, sorry for that. Also thank you for the hint/link. :)

And of course: thank you for the app! :)