OpenTracksApp/OpenTracks #623: Recording stops after one hour of screen off

ioctl-user, 2021-03-02:
Subj.

After device unlock recording starts again.

Disable battery optimization in settings doesn't help.
I have LineageOS 17.1 (Android 10) firmware.

dennisguse, 2021-03-02:
Which device?

ioctl-user, 2021-03-03:
Device Oneplus 8.

dennisguse, 2021-03-03:
Then, I guess, I cannot be very helpful.
We had some trouble with other devices, but not ones running LineageOS.
And there is not too much OpenTracks can do about - we do all the things you can do, to prevent this (basically starting a foreground service).
I would suggest to get in contact with the LineageOS team for this device.

Sorry, for not providing any other suggestions :(

ioctl-user, 2021-03-04:
Ok. Thank you for the response.

closed 2021-03-04 by ioctl-user

dennisguse, 2021-03-04:
Actually, there is one workaround (requiring a portable power supply): you could enable the "keep display on while recording"-option.
Not really efficient, but it should "do" the job.

ioctl-user, 2021-03-05:
It's too hardcore method. The first I will try calendar notifications each 45 minutes to wake up device while recording.