OpenTracksApp/OpenTracks #624: Null Pointer Exception when importing gpx file

nexx512, 2021-03-02:
**Describe the bug**
When importing gpx files through settings -> import all tracks, I get a null pointer exception.
Unfortunately I can't attach the file due to Copyright issues.
[image]

**Technical information**
 - OS: LineageOS 15.1

**Logcat**

```
<<< log_count = 21 >>>
[03-02 18:39:47.876 1395:1428 I/Adreno-EGL]
<qeglDrvAPI_eglInitialize:379>: EGL 1.4 QUALCOMM build: Nondeterministic_AU_msm8974_LA.BF.1.1.3_RB1__release_AU (I741a3d36ca)
OpenGL ES Shader Compiler Version: E031.29.00.00
Build Date: 04/04/16 Mon
Local Branch: mybranch19053788
Remote Branch: quic/LA.BF.1.1.3_rb1.12
Local Patches: NONE
Reconstruct Branch: NOTHING

[03-02 18:39:47.876 1395:1428 I/zygote]
android::hardware::configstore::V1_0::ISurfaceFlingerConfigs::hasWideColorDisplay retrieved: 0

[03-02 18:39:47.876 1395:1428 I/OpenGLRenderer]
Initialized EGL, version 1.4

[03-02 18:39:47.876 1395:1428 D/OpenGLRenderer]
Swap behavior 1

[03-02 18:39:47.884 1395:1428 W/Adreno-EGL]
<qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE

[03-02 18:39:47.887 1395:1428 W/Adreno-EGL]
<qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE

[03-02 18:39:47.897 1395:1428 D/vndksupport]
Loading /vendor/lib/hw/gralloc.msm8974.so from current namespace instead of sphal namespace.

[03-02 18:39:47.922 1395:1428 W/Adreno-EGL]
<qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE

[03-02 18:39:47.934 1395:1428 D/vndksupport]
Loading /vendor/lib/hw/gralloc.msm8974.so from current namespace instead of sphal namespace.

[03-02 18:39:49.543 1395:1428 W/Adreno-EGL]
<qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE

[03-02 18:39:49.551 1395:1428 D/vndksupport]
Loading /vendor/lib/hw/gralloc.msm8974.so from current namespace instead of sphal namespace.

[03-02 18:39:49.645 1395:1428 D/OpenGLRenderer]
endAllActiveAnimators on 0x99be8000 (RippleDrawable) with handle 0x99c0b400

[03-02 18:39:51.592 1395:1428 D/OpenGLRenderer]
endAllActiveAnimators on 0x8a7cbd00 (RippleDrawable) with handle 0x99c0bf30

[03-02 18:39:53.958 1395:1428 W/Adreno-EGL]
<qeglDrvAPI_eglGetConfigAttrib:607>: EGL_BAD_ATTRIBUTE

[03-02 18:39:53.969 1395:1428 D/vndksupport]
Loading /vendor/lib/hw/gralloc.msm8974.so from current namespace instead of sphal namespace.

[03-02 18:39:54.044 1395:9280 W/AbstractFileTrackImporter]
Marker without time ignored.

[03-02 18:39:54.051 1395:9280 I/chatty]
uid=10161(de.dennisguse.opentracks) AsyncTask #13 identical 11 lines

[03-02 18:39:54.052 1395:9280 W/AbstractFileTrackImporter]
Marker without time ignored.

[03-02 18:39:54.053 1395:9280 E/StringUtils]
Invalid XML dateTime value

[03-02 18:39:54.066 1395:9280 E/AbstractFileTrackImporter]
Unable to import file
java.lang.NullPointerException: text
	at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.createTrackPoint(AbstractFileTrackImporter.java:436)
	at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.getTrackPoint(AbstractFileTrackImporter.java:339)
	at de.dennisguse.opentracks.io.file.importer.GpxFileTrackImporter.onTrackPointEnd(GpxFileTrackImporter.java:205)
	at de.dennisguse.opentracks.io.file.importer.GpxFileTrackImporter.endElement(GpxFileTrackImporter.java:112)
	at org.apache.harmony.xml.ExpatParser.endElement(ExpatParser.java:156)
	at org.apache.harmony.xml.ExpatParser.appendBytes(Native Method)
	at org.apache.harmony.xml.ExpatParser.parseFragment(ExpatParser.java:513)
	at org.apache.harmony.xml.ExpatParser.parseDocument(ExpatParser.java:474)
	at org.apache.harmony.xml.ExpatReader.parse(ExpatReader.java:316)
	at org.apache.harmony.xml.ExpatReader.parse(ExpatReader.java:279)
	at javax.xml.parsers.SAXParser.parse(SAXParser.java:390)
	at javax.xml.parsers.SAXParser.parse(SAXParser.java:187)
	at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.importFile(AbstractFileTrackImporter.java:123)
	at de.dennisguse.opentracks.io.file.importer.GpxFileTrackImporter.importFile(GpxFileTrackImporter.java:39)
	at de.dennisguse.opentracks.io.file.importer.ImportService.importFile(ImportService.java:65)
	at de.dennisguse.opentracks.io.file.importer.ImportService.onHandleWork(ImportService.java:45)
	at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:392)
	at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:383)
	at android.os.AsyncTask$2.call(AsyncTask.java:333)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
	at java.lang.Thread.run(Thread.java:764)

[03-02 18:39:54.068 1395:9280 D/ImportService]
Parser error: java.lang.NullPointerException: text
de.dennisguse.opentracks.io.file.importer.ImportParserException: java.lang.NullPointerException: text
	at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.importFile(AbstractFileTrackImporter.java:130)
	at de.dennisguse.opentracks.io.file.importer.GpxFileTrackImporter.importFile(GpxFileTrackImporter.java:39)
	at de.dennisguse.opentracks.io.file.importer.ImportService.importFile(ImportService.java:65)
	at de.dennisguse.opentracks.io.file.importer.ImportService.onHandleWork(ImportService.java:45)
	at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:392)
	at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:383)
	at android.os.AsyncTask$2.call(AsyncTask.java:333)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
	at java.lang.Thread.run(Thread.java:764)
Caused by: java.lang.NullPointerException: text
	at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.createTrackPoint(AbstractFileTrackImporter.java:436)
	at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.getTrackPoint(AbstractFileTrackImporter.java:339)
	at de.dennisguse.opentracks.io.file.importer.GpxFileTrackImporter.onTrackPointEnd(GpxFileTrackImporter.java:205)
	at de.dennisguse.opentracks.io.file.importer.GpxFileTrackImporter.endElement(GpxFileTrackImporter.java:112)
	at org.apache.harmony.xml.ExpatParser.endElement(ExpatParser.java:156)
	at org.apache.harmony.xml.ExpatParser.appendBytes(Native Method)
	at org.apache.harmony.xml.ExpatParser.parseFragment(ExpatParser.java:513)
	at org.apache.harmony.xml.ExpatParser.parseDocument(ExpatParser.java:474)
	at org.apache.harmony.xml.ExpatReader.parse(ExpatReader.java:316)
	at org.apache.harmony.xml.ExpatReader.parse(ExpatReader.java:279)
	at javax.xml.parsers.SAXParser.parse(SAXParser.java:390)
	at javax.xml.parsers.SAXParser.parse(SAXParser.java:187)
	at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.importFile(AbstractFileTrackImporter.java:123)
	... 10 more

```

dennisguse, 2021-03-02:
The error message is very imprecise.
And here is my guess: you have an invalid (more precise missing) timestamp in your data.
Relevant info the log is ```Invalid XML dateTime value```.
And there we just re-throw the exception coming from ```DateTimeFormatter.parseBest()``` and showing this in the UI.

Not really helpful...

nexx512, 2021-03-02:
Yes I know, it lacks some important information.

Maybe I can check if I can get more information about the data in the
file.
Thanks for your info about how you parse the data.
I'm not sure about the requirements of the gpx format but are
timestamps really required here? Might it be possible that data is
still valid even without timestamps?

I haven't had the time to dig deeper into that error today. I will
check, if I can provide more info.

dennisguse, 2021-03-02:
@nexx512 I found the issue for the lack of details.
Can you test it? 
https://github.com/OpenTracksApp/OpenTracks/pull/625

nexx512, 2021-03-03:
From reviewing the code I found that https://github.com/OpenTracksApp/OpenTracks/blob/edf4ccc941d5226beae5b486238507db2e3acc50/src/main/java/de/dennisguse/opentracks/io/file/importer/AbstractFileTrackImporter.java#L434 produces a null pointer exception if there is no time tag in the track point. This leats to time being null wich causes a NPE in parseBest according to the [DateTimeFormatter Docs](https://docs.oracle.com/javase/8/docs/api/java/time/format/DateTimeFormatter.html#parseBest-java.lang.CharSequence-java.time.temporal.TemporalQuery...-). Time tags apparently are not mandatory according to https://de.wikipedia.org/wiki/GPS_Exchange_Format. So it would be nice if empty or missing time tags can be handled correctly.

I was able to reproduce the error with the gpx example given in https://de.wikipedia.org/wiki/GPS_Exchange_Format#Beispiel and attached it for further debugging.

[gpxtest.gpx.gz](https://github.com/OpenTracksApp/OpenTracks/files/6078262/gpxtest.gpx.gz)

dennisguse, 2021-03-03:
You are right, the time of a trackpoint in GPX is optional: https://www.topografix.com/GPX/1/1/#type_wptType

Then the question, what should OpenTracks do if no timestamp is available as in the internal data model the timestamp is mandatory (and assumed to be monotonically increasing).
We could ignore the trackpoint (would also ignore markers)?

nexx512, 2021-03-03:
I have a gpx file from a hiking guide that only has waypoints without time. So ignoring waypoints without timestamp would result in an empty list in my case.
If you just use the current time then this would probably cause errors if there are some waypoints with time and some without. I guess that's not a good idea. So right now I don't have a good solution to the problem.
Since your internal data model relies on timestamps, I can try to insert times in the gpx file.

Maybe someone else has a good idea how to deal with this case?

dennisguse, 2021-03-03:
@nexx512 Why do you want to import this track in the first place?

PS: We use the time to determine the speed; so it is not absolutely essential.
We could also generate something somehow.

nexx512, 2021-03-04:
My intention was to use the app as a tour map. I have ha hiking tour guide that provides gpx files for the tours. So i thougt I could use the map feature of the app as an interactive map. I'm not quite sure if the app is suitable for that purpose though. Thought I give it a shot.

dennisguse, 2021-03-04:
Interesting!
Actually, you can do that (if you get the GPX imported) as OSMDashboard supports showing multiple tracks.
1. Import the GPX
2. Start a new recording
3. Go to track list
4. Long press on one of the two tracks (triggers a multi-item selection)
5. Select the other own
6. Use the context menu (map button)
(OSMDashboard opens and shows the two tracks in different colors)

You don't get something like 'you are off the track'...
Not really the (intended) purpose, but it is just a tool ;)

@pstorch FYI

PR #625 (2021-03-06): Import: replaced SAXParserException as it only shows the message of it's cause.
Fixes #624.
closed 2021-03-06 by dennisguse