OpenTracksApp/OpenTracks #628: Speed and elevation while recording

rgmf, 2021-03-04:
**Describe the bug**
When I start a new track current speed is always 0 and elevation is "-". Anyway, all stats are computed correctly. I've not realized until today so I don't know when this problem begin (usually I start a record and save the smartphone in my pocket).

Is it reproducible for someone? It's an strange error so I don't know if it's an OT issue or a fault of my smartphone.

PS/ I cannot reproduce it in a virtual device :(

Here there is a screenshots:
[image]

**More info**
I'm not using any sensor.

**Technical information**
 - Device: Pixel XL 2
 - OS: LineageOS 17.1
 - Version: current F-Droid version (3.15.0)

dennisguse, 2021-03-04:
Now that you mention it - I have seen this behavior as well.

rgmf, 2021-03-04:
I get it but I need to check it in real scenarios (only tested on AVD)

rgmf, 2021-03-05:
`TrackDataHub` send a null TrackPoint every once in a while through `onNewTrackPointsDone`. That's why statistics recording  fragment shows 0 for speed and you sometimes feel a blink on current speed.

dennisguse, 2021-03-05:
@rgmf then I introduced the issue...
Let me fix it :)

commit 6cffd332 (2021-03-04): Recording fragment didn't show current speed and current elevation because NaN and null values. Fixes #628
PR #632 (2021-03-07): Bug: StatisticsRecordingFragment: speed and location where sometimes 0.
Fixes #628.

The TrackDataHub not send notification if no TrackPoints were processed.
commit 3dc8a724 (2021-03-07): Bug: StatisticsRecordingFragment: speed and location where sometimes 0.

Fixes #628.
closed 2021-03-07 by dennisguse