OpenTracksApp/OpenTracks #629: Start coordinates are 0.0,0.0 in KML export

voegelas, 2021-03-04:
**Describe the bug**
The start coordinates are always 0.0,0.0. There are empty gx:coord and gx:value elements in KML exports whenever a track is started, paused, resumed and stopped. OsmAnd~ shows connections to the coordinates 0,0 in the Atlantic ocean. gpsbabel fails with *kml: coord field decode failure on ""*. In JOSM, the track isn't shown as expected. Tracks from 12th Februrary, which were created with an older OpenTracks version, are fine. Is this problem related to #316?

Excerpt from an exported KML file:

```xml
<Placemark>
<!-- omitted data -->
<styleUrl>#start</styleUrl>
<Point>
<coordinates>0.0,0.0</coordinates>
</Point>
</Placemark>
```
```xml
<gx:Track>
<when>2021-03-04T12:20:47.916Z</when>
<gx:coord/>
<when>2021-03-04T12:20:48.343Z</when>
<gx:coord>8.973552 48.747633 483.2796630859375</gx:coord>
<!-- omitted data -->
<when>2021-03-04T12:48:06.926Z</when>
<gx:coord/>
<ExtendedData>
<SchemaData schemaUrl="#schema">
<gx:SimpleArrayData name="speed">
<gx:value/>
<gx:value>0.0</gx:value>
<!-- omitted data -->
<gx:value/>
</gx:SimpleArrayData>
</SchemaData>
</ExtendedData>
</gx:Track>
```

**To Reproduce**
1. Record track
2. Share track as KMZ/KML file
3. Inspect file
4. Try to convert the file with *gpsbabel -i kml -o gpx foo.kml foo.gpx*

**Technical information**
 - Device: Xiaomi Redmi 7A
 - OS: Android 10, MIUI Global 12.0.1
 - Version v3.15.1

dennisguse, 2021-03-04:
Yes, #316 is the reason for this (was released with v3.15.0: mid February).
In difference to GPX, KML/KMZ allows locations that only consists of a timestamp (and then an empty gx:coord); this is also defined in the schema.
We implemented something similar for GPX, but rolled it back as it's schema does not support this.
We export these special locations to be able to restore the TrackStatistics correctly (e.g., moving speed, moving time etc).

Which brings up the question, how to solve this?

Anyhow, #start should contain the _first real coordinates_, so that is definitely an issue.

dennisguse, 2021-03-05:
@voegelas Thanks for this detailed investigation!
I will try to fix the Placemark 0.0 issue this weekend.

dennisguse, 2021-03-05:
I just briefly looked into that issue and the actual question is: do we need placemarks for the start (#start) and end (#end) of a track?
We do not use them for the import anyhow.
My best guess is that those are/were just used to visualize it nicely in Google Earth.
And if this holds assumption holds, I am inclined to just remove them.

@rgmf @voegelas @pstorch 
Do you have any opinion on that?

PS: Simplest fix to keep the #start-placemark would be to move it to the end of the KML.

pstorch, 2021-03-05:
I don't need an extra placemark for start or end. If it is really needed for someone we could use the coordinates of the first real trackpoint (or the last for the end).

rgmf, 2021-03-06:
> I don't need an extra placemark for start or end. If it is really needed for someone we could use the coordinates of the first real trackpoint (or the last for the end).

+1

So, I'd remove them too.

voegelas, 2021-03-06:
I think that users can add start and end placemarkers manually if they need them.

[OpenLayers](https://github.com/openlayers/openlayers/blob/main/src/ol/format/KML.js) doesn't support empty gc:coord elements either. They insert 0,0 coordinates when their gxCoordParser fails to parse gc:coord elements.

closed 2021-03-08 by dennisguse

dennisguse, 2021-03-08:
Removal of #start/#end released in v3.15.3

@voegelas I created #639 as follow-up regarding interoperability with other applications.

commit 114fc825 (2021-03-08): KML export: remove #start (was broken due to #316) and #end (included track statistics) placemarks.
Fixes #629.