OpenTracksApp/OpenTracks #636: GPX export fails: File name contains invalid characters

ChristophWurst, 2021-03-06:
**Describe the bug**

I recorded a bike ride today, just like I always do. Now after updating the app I'm no longer to export the GPX to any other app. As I understand from the logcat output it looks like the encoding of spaces as %20 could be troublesome.

**To Reproduce**
1. Open a recording
2. Click the share icon
3. Select Nextcloud/K-9whatever
4. App crashes

```
03-06 18:25:38.359 23587 23616 E AndroidRuntime: FATAL EXCEPTION: AsyncTask #2
03-06 18:25:38.359 23587 23616 E AndroidRuntime: Process: de.dennisguse.opentracks, PID: 23587
03-06 18:25:38.359 23587 23616 E AndroidRuntime: java.lang.RuntimeException: An error occurred while executing doInBackground()
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at android.os.AsyncTask$4.done(AsyncTask.java:415)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:271)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:923)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: Caused by: java.lang.UnsupportedOperationException
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at de.dennisguse.opentracks.io.file.exporter.GPXTrackExporter.writeTrack(GPXTrackExporter.java:108)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at de.dennisguse.opentracks.content.provider.ShareContentProvider.lambda$openFile$1(ShareContentProvider.java:228)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at de.dennisguse.opentracks.content.provider.-$$Lambda$ShareContentProvider$UUYLBGQZWBfxU6qi91pkc6uXicw.writeDataToPipe(Unknown Source:11)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at android.content.ContentProvider$1.doInBackground(ContentProvider.java:2304)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at android.os.AsyncTask$3.call(AsyncTask.java:394)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
03-06 18:25:38.359 23587 23616 E AndroidRuntime: 	... 3 more
03-06 18:25:38.371  3401 29543 E MediaProvider: File name contains invalid characters
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: Exception while copying content://de.dennisguse.opentracks.content/tracks/gpx/46/2021-03-06%2014%3A30.gpx to temporary file
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: java.io.IOException: Operation not permitted
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at java.io.UnixFileSystem.createFileExclusively0(Native Method)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at java.io.UnixFileSystem.createFileExclusively(UnixFileSystem.java:317)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at java.io.File.createNewFile(File.java:1008)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at com.owncloud.android.ui.asynctasks.CopyAndUploadContentUrisTask.doInBackground(CopyAndUploadContentUrisTask.java:170)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at com.owncloud.android.ui.asynctasks.CopyAndUploadContentUrisTask.doInBackground(CopyAndUploadContentUrisTask.java:49)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at android.os.AsyncTask$3.call(AsyncTask.java:394)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at java.lang.Thread.run(Thread.java:923)
03-06 18:25:38.383  1618  4218 W ActivityTaskManager:   Force finishing activity de.dennisguse.opentracks/.TrackRecordedActivity
```

**Technical information**
 - Device: Pixel 3a
 - OS: Android 11
 - Version 3.15.1

dennisguse, 2021-03-06:
Related: https://github.com/OpenTracksApp/OpenTracks/issues/254

rgmf, 2021-03-06:
I can reproduce the same issue that @ChristophWurst but only if I try to share a track with NextCloud. But it works with F-9 Mail and Telegram, for example.

Also, in logcat you can see this line:
```
03-06 18:25:38.373 29187 17277 E CopyAndUploadContentUrisTask: 	at com.owncloud.android.ui.asynctasks.CopyAndUploadContentUrisTask.doInBackground(CopyAndUploadContentUrisTask.java:170)
```

Could the issue be from NextCloud?

ChristophWurst, 2021-03-06:
I renamed the activity to drop the spaces. It still errors

```
03-06 18:53:49.533 28033 28033 D ShareContentProvider: Created uri content://de.dennisguse.opentracks.content/tracks/gpx/46/2021-03-06.gpx with MIME application/gpx+xml
03-06 18:53:49.534 28033 28033 W Bundle  : Key android.intent.extra.TEXT expected ArrayList<CharSequence> but value was a java.lang.String.  The default value <null> was returned.
03-06 18:53:49.535 28033 28033 W Bundle  : Attempt to cast generated internal exception:
03-06 18:53:49.535 28033 28033 W Bundle  : java.lang.ClassCastException: java.lang.String cannot be cast to java.util.ArrayList
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.os.BaseBundle.getCharSequenceArrayList(BaseBundle.java:1343)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.os.Bundle.getCharSequenceArrayList(Bundle.java:1145)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.content.Intent.getCharSequenceArrayListExtra(Intent.java:8364)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.content.Intent.migrateExtraStreamToClipData(Intent.java:11345)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.content.Intent.migrateExtraStreamToClipData(Intent.java:11308)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1721)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.app.Activity.startActivityForResult(Activity.java:5320)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.fragment.app.FragmentActivity.startActivityForResult(FragmentActivity.java:675)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.app.Activity.startActivityForResult(Activity.java:5278)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.fragment.app.FragmentActivity.startActivityForResult(FragmentActivity.java:662)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.app.Activity.startActivity(Activity.java:5664)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.app.Activity.startActivity(Activity.java:5617)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at de.dennisguse.opentracks.TrackRecordedActivity.onOptionsItemSelected(TrackRecordedActivity.java:151)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.app.Activity.onMenuItemSelected(Activity.java:4269)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:383)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:228)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:65)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.widget.Toolbar$1.onMenuItemClick(Toolbar.java:207)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:779)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:834)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:158)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:985)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:975)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:623)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:151)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.view.View.performClick(View.java:7448)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.view.View.performClickInternal(View.java:7425)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.view.View.access$3600(View.java:810)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.view.View$PerformClick.run(View.java:28305)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.os.Handler.handleCallback(Handler.java:938)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.os.Handler.dispatchMessage(Handler.java:99)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.os.Looper.loop(Looper.java:223)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at android.app.ActivityThread.main(ActivityThread.java:7660)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at java.lang.reflect.Method.invoke(Native Method)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
03-06 18:53:49.535 28033 28033 W Bundle  : 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
03-06 18:53:49.632 28033 28050 E DatabaseUtils: Writing exception to parcel
03-06 18:53:49.632 28033 28050 E DatabaseUtils: java.lang.SecurityException: Permission Denial: reading de.dennisguse.opentracks.content.provider.ShareContentProvider uri content://de.dennisguse.opentracks.content/tracks/gpx/46/2021-03-06.gpx from pid=25153, uid=1000 requires the provider be exported, or grantUri [... cut ...]
03-06 18:53:54.063 28033 28059 I Process : Sending signal. PID: 28033 SIG: 9
03-06 18:53:54.092   953   953 I Zygote  : Process 28033 exited due to signal 9 (Killed)
03-06 18:53:54.100  1618  1640 I ActivityManager: Process de.dennisguse.opentracks (pid 28033) has died: fg  TOP 
03-06 18:53:54.102  1618  1714 I libprocessgroup: Successfully killed process cgroup uid 10288 pid 28033 in 0ms
03-06 18:53:54.108  1618  1686 W ActivityManager: setHasOverlayUi called on unknown pid: 28033
```

The tracking contains heart rate and cadence data if that is relevant.

ChristophWurst, 2021-03-06:
This could be related to https://github.com/OpenTracksApp/OpenTracks/issues/615

ChristophWurst, 2021-03-06:
> But it works with F-9 Mail

k9 doesn't crash indeed but the file sent is 0B for me.

rgmf, 2021-03-06:
> > But it works with F-9 Mail
> 
> k9 doesn't crash indeed but the file sent is 0B for me.

So, I think it's related with #615 (already fixed).

In fact, I've just tested with develop version and It works for NextCloud too.

ChristophWurst, 2021-03-06:
Thanks for checking. I see that fix went into 3.15.2. I currently get no update to that via f-droid.

I'll retest once that is available.

ChristophWurst, 2021-03-06:
Sooo how could I actually test the latest dev version? Android studio asks me about uninstalling the current app and wiping the data … that's not quite what I want. I also bought the Play version but apparently that one doesn't share data with the f-droid version :speak_no_evil:

ChristophWurst, 2021-03-06:
FYI I've reported the missing f-droid update at https://gitlab.com/fdroid/fdroiddata/-/issues/2325.

The latest release is in their data repo but somehow neither https://f-droid.org/en/packages/de.dennisguse.opentracks/ nor my phone shows the v3.15.2

dennisguse, 2021-03-06:
@ChristophWurst That is sadly "normal". For v3.15.0, it took 2 weeks.
And the changelog of v3.15.0 was already shown although only v3.14.3 was in the repo.
And when v3.15.0 went live on F-Droid it took some hours until one could actually download the APK.
Not a pleasant experience.

Some more info here: https://forum.f-droid.org/t/how-long-for-an-app-to-be-available-for-download-once-metadata-are-updated/12740

dennisguse, 2021-03-06:
v3.15.2 is live on F-Droid.
@ChristophWurst Can you check if this fixes this issue?

ChristophWurst, 2021-03-07:
It does. Thanks a lot and sorry for reporting an already fixed bug.

closed 2021-03-07 by ChristophWurst

dennisguse, 2021-03-07:
@ChristophWurst It is better to get a bug report too much than none at all.
And it is good too know that the exceptions look very different!
TLDR Thanks!