OpenTracksApp/OpenTracks #637: Support for TCX or FIT export (format that supports powermeter stats)

Ducky6-Stonework-Grumble, 2021-03-07:
Is it possible to add the file export format that support powermeter data? As I understand neither KML nor GPS are designed to store that data. Thank you.

dennisguse, 2021-03-07:
For which tool do you need this?

Ducky6-Stonework-Grumble, 2021-03-08:
Trainer road and occasionally Strava. It was a useful addition to include power meter support but and I am trying to get the fitness data extracted to use with fitness trackers.

dennisguse, 2021-03-16:
We export the power meter now to GPX: see  #638.

Implementing TCX or FIT in addition to GPX/KML is quite some effort and (for now) there is no point in doing it.
Feel free to re-open if you want to implement this.

dennisguse, 2025-03-28:
I just  checked the MyTracks implementation of TCX (I removed while initially creating OpenTracks).
My finding (after 5 years again): TCX has the same flaws like GPX.
* all trackpoints require coordinates (like GPX)
* no support for markers
* no support for images

Export+import will not restore the data identically.
This is way too complicated for the added  value for adding it to OpenTracks.

Next step: checking Garmin's FIT.

@TheSkorm FYI