OpenTracksApp/OpenTracks #639: KML with empty gx:coords.

dennisguse, 2021-03-08:
Follow-up to #629.

As of v3.15.0, we generate KML TrackPoints without coordinates (empty gx:coords).
We use this to store timestamps of events without a GPX fix (like recording started), but it will be extended for cases like #500 (indoor training without GPS).

Although it is complaint with the OGC KML 2.2/2.3 standard allows this, many open-source tools do not and use 0.0/0.0 as coordinates if none are present:
* https://github.com/OpenTracksApp/OpenTracks/issues/629#issuecomment-791555698
* https://github.com/OpenTracksApp/OpenTracks/issues/629#issuecomment-791944551

We need to talk to them, how to handle this.

Relevant part from OGC KML 2.3: http://docs.opengeospatial.org/is/12-007r2/12-007r2.html#625
```
Missing data values for positions on the track, can be accommodated by empty kml:coord(i.e. <kml:coord></kml:coord>) or kml:angles values. Any element except kml:when can be empty and may be used to represent missing data to balance the size of the kml:when, kml:coord, and kml:anglesarrays, since the data array sizes must be equal. Missing numerical data, for example represented by an empty kml:coord or kml:angles, can be estimated by linear interpolation between two well-specified data points nearby. Such interpolation can also be applied to missing numerical data arrays in kml:ExtendedData for a track.
```

voegelas, 2021-03-08:
gpsbabel and OsmAnd have already accepted pull requests. In OpenLayer's [KML.js](https://github.com/openlayers/openlayers/blob/main/src/ol/format/KML.js) the functions gxCoordParser and readGxTrack need to be adapted. If time permits, I will create a pull request this week.

dennisguse, 2021-03-08:
@voegelas awesome!

dennisguse, 2021-07-14:
Empty coordinates are now also support by OpenLayers 6.6.0: https://github.com/openlayers/openlayers/releases/tag/v6.6.0
Thanks @voegelas!

Great job!

closed 2021-07-14 by dennisguse