OpenTracksApp/OpenTracks #645: AnnouncementPeriodicTask always loads all TrackPoints for each announcement.

dennisguse, 2021-03-13:
AnnouncementPeriodicTask always loads all TrackPoints for each announcement.
The longer a track gets, the more time, we spent on loading the data.

We should somehow store the state and just load the TrackPoints that were not yet there when the last announcement was made.

Relevant code: 
```
        List<TrackPoint> trackPoints = contentProviderUtils.getTrackPoints(track.getId());
        IntervalStatistics intervalStatistics = new IntervalStatistics(trackPoints, (float) (PreferencesUtils.isMetricUnits(context) ? 1d * UnitConversions.KM_TO_M : 1d * UnitConversions.MI_TO_M));
        IntervalStatistics.Interval lastInterval = intervalStatistics.getLastInterval();
```

rgmf, 2021-04-08:
Maybe it could be improved with #703... I take this so I'll try to fix it together #703

dennisguse, 2022-05-04:
This also includes SENSOR_STATS_QUERY as it queries all the sensor data (in avg, max) for the announced track and this will scan all data as well.

commit c37a216c (2022-05-23): Announcement computes intervals from last track point. Fixes #645.
commit 5512f9a3 (2022-05-23): Announcement computes intervals from last track point. Fixes #645.
commit 3b62999f (2022-07-06): Announcement computes intervals from last track point. Fixes #645.
Bugfixed: intervals distance wasn't computed correctly.