OpenTracksApp/OpenTracks #652: GPX export TrackStatistics using TrackStatsExtension

dennisguse, 2021-03-14:
For GPX, there is an extension to export statistics about a track.
We can include those without a problem:
```
<xsd:element name="Distance" type="Meters_t" minOccurs="0"/>
<xsd:element name="TimerTime" type="Seconds_t" minOccurs="0"/>
<xsd:element name="TotalElapsedTime" type="Seconds_t" minOccurs="0"/>
<xsd:element name="MovingTime" type="Seconds_t" minOccurs="0"/>
<xsd:element name="StoppedTime" type="Seconds_t" minOccurs="0"/>
<xsd:element name="MovingSpeed" type="MetersPerSec_t" minOccurs="0"/>
<xsd:element name="MaxSpeed" type="MetersPerSec_t" minOccurs="0"/>
<xsd:element name="MaxElevation" type="Meters_t" minOccurs="0"/>
<xsd:element name="MinElevation" type="Meters_t" minOccurs="0"/>
<xsd:element name="Ascent" type="Meters_t" minOccurs="0"/>
<xsd:element name="Descent" type="Meters_t" minOccurs="0"/>
```

https://www8.garmin.com/xmlschemas/TrackStatsExtension.xsd

dennisguse, 2021-03-14:
And we could export the activity/category via TripExtension's `TransportationMode`.
https://github.com/cpesch/RouteConverter/blob/796cbc79b08f2edb1694891a6b61417b9cc104fa/gpx/src/main/doc/TripExtensionsv1.xsd#L19

akanakis, 2021-04-07:
I would like to try to fix this issue, if no one else hasn't yet.

akanakis, 2021-04-10:
I'm really sorry, but for some personal reasons I can't get involve with the project right now. So I will unassign myself from this issue and let anyone else to fix it.

dennisguse, 2021-04-10:
@akanakis Don't worry!

rgmf, 2021-04-11:
@akanakis if you can get involve in the future it would be great... we'd help you on everything you need ;)

Great to know people interesting on OpenTracks :D

akanakis, 2021-04-11:
OK thank you very much!

commit 0f370a4e (2021-12-03): GPX: export track statistics using TrackStatsExtensionv1.

Fixes #652.
commit 8945a664 (2021-12-04): GPX: export track statistics using TrackStatsExtensionv1.

Fixes #652.
commit 15408f04 (2021-12-08): GPX: export track statistics using TrackStatsExtensionv1.

Fixes #652.
commit 89f9f25c (2021-12-28): GPX: export track statistics using TrackStatsExtensionv1.

Fixes #652.
closed 2021-12-28 by dennisguse