OpenTracksApp/OpenTracks #656: No Up/downslope Data Calc

mike-wi, 2021-03-16:
neither in the Statistic Screen, nor in Interval screen. Graph Shows Elevation Data.

**Technical information**
 - Nokia 2.4
 - Android 10
-  V 3.15.4

dennisguse, 2021-03-16:
@mike-wi Can you share a screenshot?
Is it while recording or afterwards or is it something imported?

mike-wi, 2021-03-16:
It's while recording a track. 

thx -mike

[image]

[image]

[image]

dennisguse, 2021-03-16:
Actually, OpenTracks uses the barometric sensor (if present) to gather the gain/loss.
And if no sensor is present, we don't collect it resulting in the "-" for both fields.
And the Nokia 2.4 does not provide this sensor.
The GPS signal is very imprecise and would require some fancy processing.

See #619.

@mike-wi Thanks for the screenshots that was very helpful.

closed 2021-03-16 by dennisguse