OpenTracksApp/OpenTracks #657: Some ideas and feature requests

sebastianha, 2021-03-17:
Hi,

first of all THANK YOU for this great app! I really like it and having an open source app for tracking which does not sell all the data is a huge win for everybody!

I used the app for several months now to track my running and I would like to share some ideas and feature requests which came to my mind during usage. I am happy to split these into individual issues afterwards but for now I wanted not to flood the issues with small issues. Just let me know if you like some of them and I will create new issues with more details.

1 Button consistency
-----
I notices that the "summary" button is visible in the header bar on the main screen when recording. I would suggest that it should be also visible while not recording to have quick access to this feature.

2 Elevation from GPS data
-----
Currently elevation is always 0.00m for me as I do not have a barometer (I think). Wouldn't it be possible to get elevation data from the GPS? I now that this is not that exact but at least I would get an idea.

3 Make shown data configurable
-----
I would really see that I can configure the data shown on the screens.

**Example 1 -  Main screen:** Currently time and km is shown. I would be handy to also show average speed or heart rate information here. There should be enough space in the second row.

**Example 2 - Interval screen:** Currently on the right there is elevation. It would be handy to change this to e.g. heart rate information (avg/max) or other data.

4 Small screen layout
-----
I sometimes use split screen to show OsmAnd and OpenTracks both on my screen. Unfortunately the layout is not optimal for this, see screenshots

[image] [image] [image]

Perhaps it would be possible to detect the very small app layout and hide some elements of make them smaller. This would be really handy

5 OsmAnd compability as dashboard
-----
I know that you have dashboard apps to show the current track but it would be really handy to be able to use OsmAnd to show the current track of OpenTracks. I already have all my offline maps in OsmAnd and it would be wasting space to have it twice on my device. Also I sometimes use OsmAnd for navigation. It would be just nice to have the live track of OpenTracks in OsmAnd

6 Enhanced Interval Status
-----
I currently use the interval screen to check my pace but it has some drawbacks. First of all I don't see my heart rate and other information from the stats screen but also is the font too small. For me it would be optimal to see the stats for the latest interval at the top of the screen in a large font and then have the other intervals below. I created a "mockup" how this could look like:

[image]

First row: Total distance and time
Second row: Max speed and average speed (OF CURRENT INTERVAL)
Thirs row: Max and average heart rate (OF CURRENT INTERVAL)

I am not sure if this is all information, perhaps this could also be configurable. But this would definitely help me to keep track while running. Perhaps this could also be a new screen and live beside the normal interval screen.

That's all for now, again thank you for this app and let me know if I should put any of these ideas into an individual ticket!

dennisguse, 2021-03-17:
1. Yes; see https://github.com/OpenTracksApp/OpenTracks/pull/658

2. See  #619

3. Can you create a new issue for this?

4. Can you create a new issue for this?

5. Possible, but OsmAnd would need to implement our interface.
It is kind of simple, but it is not in our hands.
Ask them if they consider this.

6. Can you create a new issue for this?

PS: Way easier to keep track of things, if they are separated into individual issues.
Otherwise, discussions tend to get confusing.

dennisguse, 2021-03-17:
@sebastianha Forgot to add: all really cool ideas! Thanks!

sebastianha, 2021-03-17:
Here they are:

- https://github.com/OpenTracksApp/OpenTracks/issues/659
- https://github.com/OpenTracksApp/OpenTracks/issues/660
- https://github.com/OpenTracksApp/OpenTracks/issues/661
- https://github.com/OpenTracksApp/OpenTracks/issues/662

For 5 you are right, it's more an osmand thing.

I will close this issue, everything is in seperate issues now.

PR #658 (2021-03-20): TrackList: show aggregated stats icon in the first slot.
Fixes #657.
commit 74fa73e4 (2021-03-20): TrackList: show aggregated stats icon in the first slot.
Fixes #657.
closed 2021-03-17 by sebastianha