OpenTracksApp/OpenTracks #660: Small screen layout

sebastianha, 2021-03-17:
I sometimes use split screen to show OsmAnd and OpenTracks both on my screen. Unfortunately the layout is not optimal for this, see screenshots

[image] [image] [image]

Perhaps it would be possible to detect the very small app layout and hide some elements of make them smaller. This would be really handy.

dennisguse, 2022-05-28:
@sebastianha Does the 4.0.0 release (i.e., the UI update) solve this problem for you?

sebastianha, 2022-05-28:
As soon, as it hits f-droid, I will check.

dennisguse, 2022-05-28:
In addition, we might be able to hide some UI elements when in multi-window mode.
This method allows to get events: https://developer.android.com/reference/android/app/Activity#onMultiWindowModeChanged(boolean,%20android.content.res.Configuration)

sebastianha, 2022-05-30:
> @sebastianha Does the 4.0.0 release (i.e., the UI update) solve this problem for you?

Hi,

no, unfortunately not completely. When not running it looks much better now:

[image]

But as soon as you are on the recording screen not even one row of data is visible.

[image]

It would be nice to have one row visible, this should be do-able with hiding the top bar when the screen is below a certain height.

dennisguse, 2022-05-30:
Can you check how it looks like in Fullscreen mode? (It is in the Open tracks preferences)

sebastianha, 2022-05-30:
It looks the same. I think the is no "fullscreen" when in "split screen" mode.

rgmf, 2022-06-02:
> @rgmf In one of the screenshots, the fabButton overlaps the actual content. Any idea why this is happening?

Yes, FAB button overlap the content. In AntennaPod, for  example, they added a big padding below, so when user scroll down s-he can see the last row completely. Other solution would be to hide FAB button when scroll is on the bottom.

dennisguse, 2023-08-03:
Actually the Material3 refactoring improved this a lot. Closing for now.

closed 2023-08-03 by dennisguse