OpenTracksApp/OpenTracks #662: Enhance voice functionalty

sebastianha, 2021-03-17:
It would be really nice if the voice announcements could be configured.

a) Make interval of announcements freely configurable, down to seconds / meters. Sometime I want to keep a specific pace or heart rate and it would be nice to hear this every 30 seconds.

b) Add an option to configure what will be announced. Some ideas:
- heart rate
- average speed of current interval
- time since start
-  m or km since start
- ...

dennisguse, 2022-03-02:
Implemented in #1132.

closed 2022-03-02 by dennisguse

spacecowboy, 2022-03-02:
Half of it anyway. I have implemented some of **b** in a different branch.

[image]