OpenTracksApp/OpenTracks #665: FATAL EXCEPTION: TimerTaskExecutor

vanous, 2021-03-20:
OpenTracks crashes when Settings → Voice frequency is enabled (tested with setting to 1 and 2 minutes). 
- No other app is using audio at the same time. 
- Can replicate at any time, including outside with good GPS reception.
- Crash happens when the timer fires
- I am not sure if Recording stops, because Persistent notification is still running, GPS fix is maintained. So the background thread is still running, but UI crashes... if phone is still unlocked, it seems that after tapping "Open app again" on the "App has  stopped" dialog, app restarts, the record is there, but if the screen is locked, there is nothing left after unlock.

<details>
  <summary>Click to see my log under this fold</summary>

```
2021-03-20 10:17:17.132 25550-25550/de.dennisguse.opentracks.debug V/ElevationSumManager: elevation gain: 0.0, elevation loss: 0.0
2021-03-20 10:17:17.382 25550-25550/de.dennisguse.opentracks.debug V/ElevationSumManager: elevation gain: 0.0, elevation loss: 0.0
2021-03-20 10:17:17.555 25550-25633/de.dennisguse.opentracks.debug D/Notification: allPendingIntents
2021-03-20 10:17:17.584 25550-25633/de.dennisguse.opentracks.debug W/TrackRecordingService: Ignore insertTrackPoint. trackPoint time same as last valid track point time.
2021-03-20 10:17:17.585 25550-25633/de.dennisguse.opentracks.debug D/ElevationSumManager: Reset
2021-03-20 10:17:17.594 25550-25633/de.dennisguse.opentracks.debug D/TrackStatisticsUpdater: TrackStatisticsUpdater{trackStatistics=TrackStatistics { Start Time: 2021-03-20T09:16:16.978Z; Stop Time: 2021-03-20T09:16:16.978Z; Total Distance: 0.0; Total Time: PT0S; Moving Time: PT0S; Max Speed: 0.0; Min Elevation: Infinity; Max Elevation: -Infinity; Elevation Gain: null; Elevation Loss: null}}
    
    --------- beginning of crash
2021-03-20 10:17:17.611 25550-25626/de.dennisguse.opentracks.debug E/AndroidRuntime: FATAL EXCEPTION: TimerTaskExecutor
    Process: de.dennisguse.opentracks.debug, PID: 25550
    java.lang.NullPointerException: Attempt to invoke interface method 'long android.content.SharedPreferences.getLong(java.lang.String, long)' on a null object reference
        at de.dennisguse.opentracks.util.PreferencesUtils.getRecordingTrackId(PreferencesUtils.java:57)
        at de.dennisguse.opentracks.util.PreferencesUtils.getRecordingTrackId(PreferencesUtils.java:62)
        at de.dennisguse.opentracks.services.tasks.AnnouncementPeriodicTask.announce(AnnouncementPeriodicTask.java:173)
        at de.dennisguse.opentracks.services.tasks.AnnouncementPeriodicTask.run(AnnouncementPeriodicTask.java:141)
        at de.dennisguse.opentracks.services.tasks.TimerTaskExecutor$1.run(TimerTaskExecutor.java:63)
        at java.util.TimerThread.mainLoop(Timer.java:555)
        at java.util.TimerThread.run(Timer.java:505)
2021-03-20 10:17:17.662 25550-25550/de.dennisguse.opentracks.debug V/ElevationSumManager: elevation gain: 0.0, elevation loss: 0.0
2021-03-20 10:17:17.703 25550-25550/de.dennisguse.opentracks.debug D/ViewRootImpl@29c72ee[TrackRecordingActivity]: MSG_WINDOW_FOCUS_CHANGED 0
2021-03-20 10:17:17.710 25550-25550/de.dennisguse.opentracks.debug I/TrackRecordingServiceConnection: Binding the service.
2021-03-20 10:17:17.777 25550-25550/de.dennisguse.opentracks.debug V/Surface: sf_framedrop debug : 0x4f4c, game : false, logging : 0
2021-03-20 10:17:17.778 25550-25550/de.dennisguse.opentracks.debug D/ViewRootImpl@4d7fbbf[TrackListActivity]: Relayout returned: old=[0,0][720,1280] new=[0,0][720,1280] result=0x7 surface={valid=true 505201262592} changed=true
2021-03-20 10:17:17.782 25550-25586/de.dennisguse.opentracks.debug D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000,  [720x1280]-format:1
2021-03-20 10:17:17.782 25550-25586/de.dennisguse.opentracks.debug D/OpenGLRenderer: eglCreateWindowSurface = 0x75b522a2b0
2021-03-20 10:17:17.786 25550-25550/de.dennisguse.opentracks.debug I/TrackRecordingServiceConnection: Connected to the service.
```
</details>

**Technical information**
 - Device: Samsung Galaxy S7 Edge 
 - OS: Android 8.0
 - OT Version: current master 997ab6060

dennisguse, 2021-03-20:
@vanous Can you check #666?

Don't know yet why this happens and cannot test it due to lack of TTS on my devices.

vanous, 2021-03-20:
Fix confirmed, issue solved, thank you @dennisguse !

closed 2021-03-20 by vanous

dennisguse, 2021-03-20:
@vanous thanks for testing and finding this bug!

PR #666 (2021-03-20): AnnouncementPeriodicTask: acquire sharedprefences on start (so it can…
… be restarted).

Introduced in 15fbd90ca4579d696d1e852bf10c4ecea3321736.
Fixes #665.
commit 0a006d64 (2021-03-20): AnnouncementPeriodicTask: acquire sharedprefences on start (so it can be restarted).

Introduced in 15fbd90ca4579d696d1e852bf10c4ecea3321736.
Fixes #665.