OpenTracksApp/OpenTracks #667: Delay after pressing GPS start button on main screen

sebastianha, 2021-03-20:
**Describe the bug**
When pressing the GPS start button on the main screen to prepare a GPS fix it takes several (~5s, but sometimes longer) seconds before the location icon from Android OS appears and the device is searching for a fix. In other applications the GPS icon in the tray appears immediately.

**To Reproduce**
1. Go to main screen
2. Click on GPS start button
3. See Android header bar, GPS icon does not appear immediately

If applicable:
[image]

**Technical information**
 - Device: Moto G7 Power
 - OS: LineageOS 17.1
 - Version: 3.15.4 but also has been before

dennisguse, 2021-03-20:
I am not sure that we are not responsible for the icon.
I presume that you are referencing the one beside the _airplane mode icon_, right?
I assume that this is feedback from Android showing the user that the phone got a GPS fix as some app was requesting it (might be in the background).

Only info I found in 2min searching was this: https://consumer.huawei.com/en/support/content/en-us00407938/

@sebastianha Nice issue reports! Very detailed and easy to follow!

sebastianha, 2021-03-21:
1. Yes, it's an Android thing
2. Correct, It's about the Android native "location accessed" icon
3. The Icon should appear as soon as an app requests the current location and it does not matter if the GPS has a fix or not.

Interestingly other apps (e.g. GPSTest or OsmAnd) show the location icon immediately after starting. Opentracks does not. That's what is wondering me. I also have the feeling the OpenTracks somehow does not start the "location search" process.

This behaviour seems to happen when I have no GPS reception at all. I did a test in the basement and recorded a video:

https://user-images.githubusercontent.com/420807/111916644-2b2ed780-8a7c-11eb-84c9-735c6319abca.mp4

What you see on the video (recorded in the basement with no GPS reception)
1. No GPS icon, starting GPSTest, icon is triggered right away
2. Closing GPSTest, icon disappears
3. Starting GPSTest again, icon again triggered right after start
4. Closing GPSTest, icon disappears
5. Starting OpenTracks, hitting "Start GPS" in OpenTracks, no GPS icon in header bar.
6. Repeated test with OpenTracks
7. Test against GPSTest, icon is triggered right away

Somehow GPSTest is requesting the location in another way so the GPS icon is shown even while not having a fix. With good reception the icon appears as soon as I get a fix with OpenTracks.

I find this behaviour misleading because the first times I had the feeling that the GPS did not start after hitting the "Get GPS button" in Opentracks. I am not sure if this has any other effects we might not know.

Might not be a bug but at least got me thinking what happens here. Perhaps the location fix could be sped up by using the same method as GPSTest does, I don't know. In my perception OpenTracks takes longer to get a valid fix.

Just for reference: I do not use any other location provider (google, wifi, bt, cellular) only plain GPS everything else is disabled.

dennisguse, 2021-04-05:
I just briefly checked the GPSTest code (mainly GpsTestActivity).
They are registering multiple listeners for the LocationManager besides locations (e.g., antenna info, NMEA, GnssMeasurementsEvent).
I guess one of these additional things trigger the icon as the GPS hardware is accessed and starts doing something.

All these OpenTracks doesn't use - it only registers via `requestLocationUpdates`.
So, my best guess is that the icon appears, _when_ the first data is provided to OpenTracks - we might still dismiss the data as the provided accuracy is not (yet) sufficient.

@sebastianha Please re-open if this is an still an issue for you.

closed 2021-04-05 by dennisguse

sebastianha, 2021-04-06:
Sounds reasonable, thanks for looking into it!