OpenTracksApp/OpenTracks #671: Provide a way to start tracking from an Automate/Tasker flow

chmduquesne, 2021-03-21:
**Is your feature request related to a problem? Please describe.**
I always forget to start tracking my sport activities, and as a result I am missing the gps tracks of glorious hikes. I would like to program my phone to automatically record my gps track when the right conditions are met.

**Describe the solution you'd like**
I would like to start recording from the App automate. This way, I can be fully in charge of determining exactly when the start/stop recording actions should be started. In my case, I would use my gps location, provided with a change of personal activity.

**Describe alternatives you've considered**
I have considered using strava, which allows to do that via the data uri `http://strava.com/nfc/record`. You can also provide the ride type in an Extras json object.

dennisguse, 2021-03-21:
I guess you want an application link, right?
https://developer.android.com/training/app-links
There seems to be some web-based configuration required: https://developer.android.com/training/app-links#add-app-links

Could we also handle an intent instead?
I don't know what Automate/Tasker supports.

chmduquesne, 2021-03-22:
Yes, it is also possible to send an intent. Both Automate and Tasker can send arbitrary broadcast intents:

* https://llamalab.com/automate/doc/block/broadcast_send.html
* https://tasker.joaoapps.com/userguide/en/intents.html

Tasker documentation is clear about the limitations (and I believe Automate has the same limitations):

> Limitations
> * Tasker can only receive intents which are sent to broadcast receiver components, not to activities or services.
> * some intent senders require that a corresponding intent filter is specified statically (i.e. in an Android Manifest). Those intents cannot be received.
> * intents which are broadcast with a specification of a particular package component to receive it cannot be received.

chmduquesne, 2021-03-22:
> I guess you want an application link, right?

I am not picky about the way it should be implemented, whatever is the most convenient. Any way that would allow me to start/stop tracking from a flow would make me happy 🙂

arjan-s, 2021-09-15:
I second this feature request! I've worked on Fossil Hybrid HR support in [Gadgetbridge](https://www.gadgetbridge.org/), and would love to add support for GPS tracking starting/stopping from my watch. Intents seem to be the way to go, both for my use case and the original one above.

dennisguse, 2021-09-15:
@arjan-s go ahead.
One limitation: TrackRecordingService should not be exported to other applications.

Current implementation: TrackRecordingService is started and OpenTracks activities bind to this service and start/pause/stop recordings.

Anti-goals: expose any information about OpenTracks state (e.g., is a recording running) to any other application.

We may need to add another component that is exported and can  start/ stop TrackRecordingService.

Open points: 
- do we start any UI (beside the service notification)?
- we may need to define an intent action?

dennisguse, 2021-09-15:
We may be able to use TrackRecordingActivity to start recordings via external intent - almost everything is in place for this.

Another anti-goal: resuming a track.

dennisguse, 2021-12-30:
@arjan-s @chmduquesne Do you have time to test this?
https://github.com/OpenTracksApp/OpenTracks/pull/1072

chmduquesne, 2021-12-30:
@dennisguse is the feature is available on a prebuilt version of the app?

dennisguse, 2021-12-30:
@chmduquesne Yes, there is a build for each branch.
Can be downloaded here: https://github.com/OpenTracksApp/OpenTracks/actions/runs/1637768469

(just search for APK).

arjan-s, 2021-12-30:
@dennisguse Hey, that's great! I can test it in a few days probably. Are you planning to add a `STOP_RECORDING` intent too? My goal is to allow certain wearables to start/stop OpenTracks recording via Gadgetbridge, but that would obviously require an intent to stop recording as well.

dennisguse, 2021-12-30:
Adding stop, won't be a problem.

Only limitations: we should not expose any information about OpenTracks current state.

@arjan-s If START_RECORDING works, STOP_RECORDING is just 2seconds copy and paste :)

arjan-s, 2021-12-30:
@dennisguse That would be great!! For my use case there isn't a real need for feedback. Just the possibility to start and stop recording would already be awesome to have!

Now I just need some time to write down all my settings, install the OpenTracks test build, and reselect all settings. :)

arjan-s, 2022-01-01:
@dennisguse I installed the linked nightly, which turned out to have a different package I'd, so I didn't have to uninstall the F-Droid version. 👍🏻
I didn't get  the intent to work though. I used Tasker to send it, which works fine for other apps. Changing the intent action to the nightly package name didn't work either. Any idea what I might be doing wrong?

dennisguse, 2022-01-01:
I tested it with Easer.
Best guess: I guess you used the wrong nightly (we have two: the F-Droid one and the Github one; while only the later is built per branch).

[image]

Here is a PlayStore build of this branch.
[OpenTracksUpstream-releasePlayStore.zip](https://github.com/OpenTracksApp/OpenTracks/files/7797920/OpenTracksUpstream-releasePlayStore.zip)

arjan-s, 2022-01-01:
Interesting... I apparently did the correct thing, but it simply has no effect. I'm sending the correct intent (tried with both Tasker and Easer), but I get no response at all from OpenTracks. Also uninstalled the official (f-droid) OpenTracks so I just have the nightly running. Btw I was running the correct GitHub nightly, but I just tried your new Play Store build and it doesn't do anything either.

arjan-s, 2022-01-01:
Tried running OpenTracks directly from source in debug mode with Android Studio, but still nothing. No logging at all about any received intent. Then I figured I could send the intent with `adb` as well, and that finally gave me some output:
```
$ adb shell am start-activity -a de.dennisguse.opentracks.publicapi.START_RECORDING
Starting: Intent { act=de.dennisguse.opentracks.publicapi.START_RECORDING }
Error: Activity not started, unable to resolve Intent { act=de.dennisguse.opentracks.publicapi.START_RECORDING flg=0x10000000 }
$
```
Still no clue why, but apparently the intent is not registered in Android somehow.

dennisguse, 2022-01-01:
"Interesting" as it should be working.

https://github.com/OpenTracksApp/OpenTracks/blob/a5802bb24f3bb98bd853a3ce1100d1b013c12c73/src/main/AndroidManifest.xml#L77

I will try tomorrow with my 2nd device.

arjan-s, 2022-01-01:
Yeah, I don't understand it either. The manifest seems to be correct (although I'm not a very experienced Android developer).
If it helps, I'm running it on Android 11 from Samsung.

dennisguse, 2022-01-01:
I am running LineageOS 18.1 / Android 11 :D

PS/ me neither :sunglasses:

arjan-s, 2022-01-01:
Found it! Thanks to [this answer](https://stackoverflow.com/a/52110024). Add the following in the `<intent-filter>` to make it work:
```
<category android:name="android.intent.category.DEFAULT" />
```
I do have 3 issues with the current implementation though, of which the first two are blocking for my Gadgetbridge plans (sorry...):

1. Can't stop recording with an intent (already said this before and it's a todo in the PR I see)
2. Screen has to be on and unlocked for the recording to start (makes this useless for starting recording from a wearable using Gadgetbridge)
3. Currently no way to select activity type when using the intent

dennisguse, 2022-01-01:
+1 

Don't know how we can handle the 2nd: we may use an intermediate service that does all the handling and not an activity.

closed 2022-01-23 by dennisguse

kulbhushanchand, 2022-02-11:
Hi @dennisguse 
Thanks for this awesome app.
Somehow, I am unable to start the track recording by sending intent from the tasker.
I think may be I'm not using tasker correctly, so I decided to ask you here instead of creating a new issue.
I'm using the latest FDroid `version 3.24.1` and have enabled the api option in OpenTracks.
In the "Send Intent" action in tasker - 
```
Action: de.dennisguse.opentracks.publicapi.StartRecording
Cat: Default
Package: de.dennisguse.opentracks
Target: Activity
```
Nothing happens when running the task.

kulbhushanchand, 2022-02-11:
Thanks a ton @arjan-s 🙏
Working flawlessly now...

dennisguse, 2022-02-11:
@arjan-s would you have the time to do this?
I am not a user of this, so I don't know exactly what people are looking for :man_shrugging:

arjan-s, 2022-02-11:
@dennisguse yeah I figured you would ask that. :)
No problem! Expect it in a few days.

arjan-s, 2022-02-16:
For future reference: I corrected the text in the README in https://github.com/OpenTracksApp/OpenTracks/pull/1125

commit 31b98586 (2022-01-11): Public API: Public API: enable other applications to start recordings via a broadcast receiver.

Fixes #671.
commit d845e4ae (2022-01-11): Public API: enable other applications to start recordings.

Fixes #671.
commit 03d968d0 (2022-01-11): Public API: enable other applications to start recordings.

Fixes #671.
commit b9818189 (2022-01-19): Public API: enable other applications to start recordings.

Fixes #671.