OpenTracksApp/OpenTracks #681: KmzTrackImport Context.

rgmf, 2021-03-25:
KmzTrackImport has a Context like a member but some of its method receive the context like argument. Does it have any sense? Why not to use the Context on these methods?

The constructor:

```java
public KmzTrackImporter(Context context, Uri uriFile) {
        this.context = context;
        this.uriKmzFile = uriFile;
    }
```

One of its method (for example):

```java
private void cleanImport(Context context, Track.Id trackId) {
```

dennisguse, 2021-03-25:
Best guess: some leftovers that were forgotten and it is actually confusing as the parameter actually suggests that there could be another context.

dennisguse, 2021-04-04:
Fixed in e09b04308c5d62173458e7f803d39a2ba33bab91

closed 2021-04-04 by dennisguse