OpenTracksApp/OpenTracks #685: Unable to uninstall the app

deepakshilkar, 2021-03-30:
**Describe the bug**
I am unable to uninstall the app through both F-Droid as well as the pop-up pill menu. On clicking uninstall, the package installer crashes.

**To Reproduce**
1. Just long press on the app icon and uninstall. 
2. Or go to F-Droid and click uninstall. 

If applicable:

[image]

**Technical information**
 - Device: Redmi Note 5 Pro whyred
 - OS: LineageOS 17.1
 - Version v3.15.4

dennisguse, 2021-03-30:
Can you share a logcat? 

We had something very similar: https://github.com/OpenTracksApp/OpenTracks/issues/129
Solution there was to use `adb` to install as the package installer has a bug in some Android versions.

dennisguse, 2021-04-02:
I guess, we cannot do anything about this.

PR #771 (2021-05-16): Disable hasFragileUserData for production.
Should fix #685 and #129.

`hasFragileUserData=true` allows user to uninstall OpenTracks without deleting the app-internal data (actually the user can choose).

__Problem 1:__ Seemingly `hasFragileUserData=true` prevents uninstalling OpenTracks (best guess: some bugs in older Android variants).

__Problem 2:__ Anyhow: `hasFragileUserData=true` is actually an anti-feature from a privacy perspective.
It is likely that a user uninstalled OpenTracks without deleting the app-internal data.
So, the data might stay on the device and the user might not be awa [... cut]
commit 0e3ffa58 (2021-05-16): Disable hasFragileUserData for production.

Should fix #685 and #129.
closed 2021-04-02 by dennisguse