OpenTracksApp/OpenTracks #686: Introduce quick links between pages

sebastianha, 2021-03-30:
**Is your feature request related to a problem? Please describe.**
There are lists on some screen which represent data. Currently a tap action is recognized (list entry gets darker) but there is no action triggered.

**Describe the solution you'd like**
I have two examples where a tap on a list entry could lead to another page and show data linked to the list entry.

1. Aggregated Stats -> Filtered List
When clicking on an activity on the aggregated stats page this could lead back to the list with the activities but filter by the clicked activity
[image]

2. Interval List -> Chart
When clicking on an entry of the interval list this could open the chart page and change the range of the chart fitting to the selected interval
[image]

dennisguse, 2021-03-30:
How should the back button behave in this case?

sebastianha, 2021-03-30:
I would suggest to go back to the screen  on the left.

closed 2024-12-29 by dennisguse