OpenTracksApp/OpenTracks #688: Can't connect with Amazfit Neo

racuna, 2021-03-31:
**Is your feature request related to a problem? 
I have Gadgetbridge connected to Amazfit Neo with "3rd party realtime HR access" enabled. I opened OpenTracks settings, but in Bluetooth can't detect the watch. Am I missing something?

**Describe the solution you'd like**
Amazfit Neo doesn't have GPS, I wanted to merge data from Gadgetbridge (Steps, HR) with data from OpenTracks (GPS data)

**Describe alternatives you've considered**
Maybe there's a python script to do that after every workout and generate some TCX or a CSV with all the data merged.

**Additional context**
I want to ditch official Zepp app, and looks like a combination of GadgetBridge with OpenTracks will be the sweet spot

dennisguse, 2021-03-31:
It might be that you need to start the HR reading on the device first.

And then...
We tested the 3rd party realtime HR feature with a Miband3 and there the issue was that the device is not announcing it's sensor in its Bluetooth discovery responses. When asking it for HR it works. So, we added a workaround for Mibands.
We might need to do the same for the Amazfit Neo.
Could you check how the discovery response looks and if it contains the HR service?

racuna, 2021-04-01:
Is MIBand3 workaround compatible with MI Band 4? because I now tested with the Mi Band 4 of my wife and detected it pretty quick.

In the official Zepp app I enabled "Discoverable" and "Activity heart rate sharing", that worked :) thanks.

dennisguse, 2021-04-01:
The Miband4 might be working as well - but it was never tested by me.

The workaround is here: https://github.com/OpenTracksApp/OpenTracks/blob/ed5da55fbad4a87cb2e16d70029d3c0f9edbb621/src/main/java/de/dennisguse/opentracks/util/BluetoothUtils.java#L45
All devices that provide one of these service UUIDs in their discovery response are shown in the OpenTracks' settings dialog.

Here are some more details: https://github.com/OpenTracksApp/OpenTracks/issues/447#issuecomment-739239070

racuna, 2021-04-01:
Thanks, this will detect steps as well? (to calculate running cadence)

dennisguse, 2021-04-01:
@racuna Nope, OpenTracks does not use step data.
It wouldn't be too complicated to built, but so far there was no real need.
And again: lack of testing devices.

We have an open issue for this: https://github.com/OpenTracksApp/OpenTracks/issues/463
But I am not sure, if we can get the step data from a smartwatch like the Neo.

racuna, 2021-04-01:
> OpenTracks does not use step data.

Fair enough.

> but so far there was no real need.

just for nerdy running stuff https://www.polar.com/blog/what-is-running-cadence/

My brother is like 10x better at running than me, and he never cared about running technique, breathing, cadence and so on. He just runs.

dennisguse, 2021-04-06:
@racuna Could you actually use the HR from the Neo with OpenTracks?

racuna, 2021-04-06:
Yes, today I went for a run and OpenTracks was able to read the heart rate without any problems. 

The only thing was that OpenTracks only recorded last 400 meters, but maybe that was my fault because I forgot to disable the MIUI battery "optimizations" after installed the app.

But next time I will tell.

dennisguse, 2021-04-06:
@racuna Nice!
Can you add the info about that the HR works for the Neo here: https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md

closed 2021-04-06 by dennisguse