OpenTracksApp/OpenTracks #696: Crash in ChartFragment

dennisguse, 2021-04-03:
Found while testing #680, but not related to changes in #680.

I don't know how to exactly reproduce this failure, but _I guess_ that the ChartFragment is paused and then resumed.
The resume will trigger loading the data via TrackDataHub, but processing might not take place as all methods check `if (isResumed())`.
And therefore `clearTrackPoints()` might not be executing the data cleanup in ChartFragment.
And then we try to add old TrackPoints _again_ to the TrackStatistics.
And may lead to the crash.

Feels like a race condition.

```
2021-04-03 17:51:29.373 18724-19125/? E/AndroidRuntime: FATAL EXCEPTION: TrackDataHub
    Process: de.dennisguse.opentracks.debug, PID: 18724
    java.lang.RuntimeException: stopTime cannot be less than startTime: 2021-04-03T15:51:20.379Z 2021-04-03T15:33:48.631Z
        at de.dennisguse.opentracks.stats.TrackStatistics.setStopTime(TrackStatistics.java:157)
        at de.dennisguse.opentracks.stats.TrackStatisticsUpdater.internalAddTrackPoint(TrackStatisticsUpdater.java:125)
        at de.dennisguse.opentracks.stats.TrackStatisticsUpdater.addTrackPoint(TrackStatisticsUpdater.java:110)
        at de.dennisguse.opentracks.fragments.ChartFragment.createPendingPoint(ChartFragment.java:326)
        at de.dennisguse.opentracks.fragments.ChartFragment.onSampledInTrackPoint(ChartFragment.java:233)
        at de.dennisguse.opentracks.content.TrackDataHub.notifyTrackPointsTableUpdate(TrackDataHub.java:390)
        at de.dennisguse.opentracks.content.TrackDataHub.loadDataForAll(TrackDataHub.java:246)
        at de.dennisguse.opentracks.content.TrackDataHub.lambda$start$1$TrackDataHub(TrackDataHub.java:156)
        at de.dennisguse.opentracks.content.-$$Lambda$TrackDataHub$TZP5DuLlUVB-kErcp3WpHgOwuus.run(Unknown Source:2)
```

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: [e.g. Nexus 7 (2013)]
 - OS: [e.g. LineageOS 15.1]
 - Version [e.g. v3.2.4]

rgmf, 2021-04-11:
Same crash on IntervalStatistics:

```
Process: de.dennisguse.opentracks.debug, PID: 10206
    java.lang.RuntimeException: stopTime cannot be less than startTime: 2021-04-11T09:49:33.475Z 2021-04-11T09:03:50.509Z
        at de.dennisguse.opentracks.stats.TrackStatistics.setStopTime(TrackStatistics.java:155)
        at de.dennisguse.opentracks.stats.TrackStatisticsUpdater.internalAddTrackPoint(TrackStatisticsUpdater.java:122)
        at de.dennisguse.opentracks.stats.TrackStatisticsUpdater.addTrackPoint(TrackStatisticsUpdater.java:107)
        at de.dennisguse.opentracks.viewmodels.IntervalStatistics.<init>(IntervalStatistics.java:35)
        at de.dennisguse.opentracks.viewmodels.IntervalStatisticsModel.loadIntervalStatistics(IntervalStatisticsModel.java:50)
        at de.dennisguse.opentracks.viewmodels.IntervalStatisticsModel.upload(IntervalStatisticsModel.java:81)
        at de.dennisguse.opentracks.fragments.IntervalsFragment.uploadIntervals(IntervalsFragment.java:187)
        at de.dennisguse.opentracks.fragments.IntervalsFragment.lambda$new$0$IntervalsFragment(IntervalsFragment.java:62)
        at de.dennisguse.opentracks.fragments.-$$Lambda$IntervalsFragment$M9LXp0uwukRCKmOfCjZCpsVoAvc.onSharedPreferenceChanged(Unknown Source:2)
        at de.dennisguse.opentracks.fragments.IntervalsFragment.onResume(IntervalsFragment.java:143)
```

So, I guess the bug is on TrackStatistics/TrackStatisticsUpdater, not in ChartFragment.

dennisguse, 2021-04-11:
My current hypothesis is that there is a race condition in onResume.
In both cases (Chart+Interval), the TrackDataHub is resumed (triggering a complete reload) and the fragments ignore the `clearTrackPoints()` callback from TrackDataHub (to reset their TrackStatistics and compute it again).
i guess they ignore it, because the Fragments are sometimes not yet resumed while their parent activity already is (and thus reloaded the TrackDataHub).

I guess, the isResumed() check is the issue in both fragments:
```
    @Override
    public void clearTrackPoints() {
        if (isResumed()) {
            viewModel.clear();
        }
    }
```

rgmf, 2021-05-18:
I got the same error importing a track recorded with other device in OpenTracks.

Reading the KML file I see this:
```xml
...
<gx:MultiTrack>
    <gx:Track>
        <when>2021-05-16T06:10:25.029Z</when>
        <gx:coord/>
        <when>2021-05-16T06:10:25.629Z</when>
        <gx:coord>-0.80604 38.158524 65.0</gx:coord>
        <when>2021-05-16T06:10:26.597Z</when>
...
        <gx:coord>-0.801607 38.160851 64.0</gx:coord>
        <when>2021-05-16T09:12:25.601Z</when> <!-- START TIME -->
...
    </gx:Track>
    <gx:Track>
        <when>2021-05-16T09:12:25.587Z</when> <!-- END TIME IS BEFORE START TIME -->
        <gx:coord>-0.801609 38.160853 64.0</gx:coord>
    </gx:Track>
</gx:MultiTrack>
```

dennisguse, 2021-05-19:
@rgmf This crash cannot happen anymore in ChartFragment as the TrackDataHub nowadays computes the TrackStatistics.
And also for IntervalStatistics I don't see the potential for this crash (anymore).

... but I am afraid this https://github.com/OpenTracksApp/OpenTracks/issues/696#issuecomment-843149218
Should we move this to a new issue?

rgmf, 2021-05-19:
> @rgmf This crash cannot happen anymore in ChartFragment as the TrackDataHub nowadays computes the TrackStatistics.
> And also for IntervalStatistics I don't see the potential for this crash (anymore).
> 
> ... but I am afraid of this [#696 (comment)](https://github.com/OpenTracksApp/OpenTracks/issues/696#issuecomment-843149218)
> Should we move this to a new issue?

I think it's related because that kml file is from a track that crashed when I was biking. In other way it's quite difficult to reproduce.

I always export from a device to another device so I'll be careful of that, it's not necessary to move it. I hope it's fixed too.

dennisguse, 2021-05-26:
Actually fixed.

closed 2021-05-26 by dennisguse