OpenTracksApp/OpenTracks #698: Activity icon

midros, 2021-04-06:
Is there any way to add a new icon when you add a new activity?

I have added a new kayaking activity and I would really like a new icon for this type of activity

pstorch, 2021-04-06:
@midros do you have an icon already or do you know an existing icon which we can use or derive from?

midros, 2021-04-06:
@pstorch 
I don't have one but I can look for one if you like.
What about the existing icons? Is there a certain icon pack used or somebody created them from scratch?

pstorch, 2021-04-06:
Some were already there from the previous project Google MyTracks, right @dennisguse?
Some newer ones were derived from them.

@midros if you have a good icon, we can use as a template, it would be nice. I can try to fit it into OpenTracks.
For me it is a bit difficult to distinguish icons for kayaking and similar looking ones like canoeing.

midros, 2021-04-06:
@pstorch 
Nobody needs different icons for canoeing and kayaking (I hope)

I 'll try to find one similar to the ones already used and keep you informed

thanks a lot

midros, 2021-04-06:
@pstorch 
maybe [this ](https://thenounproject.com/term/kayak/17483/) or [this ](https://thenounproject.com/term/kayak/552/) will do the trick?
they are public, with several sizes and fit nicely to the rest of the icons

pstorch, 2021-04-06:
Thanks, I'll have a look

pstorch, 2021-04-06:
1.)

[image]

2.)

[image]

Which do you prefer? I think I like the second a little bit more.

dennisguse, 2021-04-06:
Nr. 2 as it is also "flat" aka 2D aka non-3D :)

pstorch, 2021-04-06:
Only thing with No2: it has a helmet or a haircut. I can try to make a version with only a circle head, so it looks like the others.

dennisguse, 2021-04-06:
@pstorch I did not even notice this ;)

pstorch, 2021-04-06:
> @pstorch I did not even notice this ;)

Yes, on a small screen it's barely visible.

Now without helmet:

[image]

midros, 2021-04-06:
I am impressed!

great

rgmf, 2021-04-06:
@pstorch it's great!!!

commit 8713eb3d (2021-04-07): fix #698 added activity icon for kayaking
closed 2021-04-07 by pstorch