OpenTracksApp/OpenTracks #703: Intervals distance don't match with stats

rgmf, 2021-04-07:
**Describe the bug**
The addition of all intervals don't match with total distance in stats.

I realized of this while I was testing #701 but it's not related with it. I guess this issue is because intervals are not considering the new types of track points (TrackPoint.Type).

I think the addition of all intervals were nearly the same that total distance in stats, weren't they?

I'm going to check it out.

**To Reproduce**
1. Record a new track.
2. Look at stats distance and intervals distance: they not match.

dennisguse, 2021-04-07:
Don't know ;)

Why not use the TrackStastistics via TrackStatisticsUpdater (similar to the charts)?
Then all computation/presentation is done this combination and we are consistently correct/wrong :)

rgmf, 2021-04-07:
> Don't know ;)
> 
> Why not use the TrackStastistics via TrackStatisticsUpdater (similar to the charts)?
> Then all computation/presentation is done this combination and we are consistently correct/wrong :)

Yes, I was thinking on it.

It has should been done like this... I'm going to try to do it like this.

dennisguse, 2021-04-07:
Is this related with the IntervalStatistics vs TrackStatistics?
If yes, then this "imprecision" also affects the voice announcements, right?

rgmf, 2021-04-07:
> Is this related with the IntervalStatistics vs TrackStatistics?
> If yes, then this "imprecision" also affects the voice announcements, right?

Yes, I think so. We are using IntervalStatistics for getting the current interval data to announce them.

rgmf, 2021-04-07:
It's strange because the distance should be exactly the same and I'm fairly sure that one day it was :thinking: 

Anyway, it's a good moment to use TrackStatistics via TrackStatisticsUpdater :smile:

rgmf, 2021-04-09:
@dennisguse  I've fixed this... but:

I don't understand very well an important detail. I have a very little track:
- 0.95 km
- 46 seconds

[image]

But those 46 seconds should be 49 seconds:
- First trackPoint (from database) has a time of: 1617571177713
- Last trackPoint (from database) has a time of: 1617571227232

So `1617571227232 - 1617571177713 = 49519 millis`

Is there an explanation on this? Could this be a potencial bug?

PS/ All trackPoints type are 0 (unless the first one, -2, and the last one, 1).

dennisguse, 2021-04-09:
@rgmf That was fast!

This issue sounds like a bug.
Can you check what happens if you export and re-import the track?
It might be that the duration computation is slightly (still) slightly different while recording.

rgmf, 2021-04-09:
> @rgmf That was fast!
> 
> This issue sounds like a bug.
> Can you check what happens if you export and re-import the track?
> It might be that the duration computation is slightly (still) slightly different while recording.

After export/import the track has:
- Total moving time: 46 seconds.
- Total time: 49 seconds.

[image]

So intervals computation through TrackStatisticsUpdater seem okay.

Intervals distance match with track stats imported/exported. There are always slightly differences between time and distance for tracks when they are exported/imported with the original recorded ones.

For example: ("una imagen vale más que mil palabras" :P ):

[image]

dennisguse, 2021-04-09:
So, we have same trouble in the TrackRecordingService, as there shouldn't be a difference (at least not larger than rounding issues)...

@rgmf Lo mismo motivo funciona en Aleman :)

dennisguse, 2021-05-06:
@rgmf This is fixed, right?

rgmf, 2021-05-06:
@dennisguse no, it's not. The potential bug looks like is in TrackRecordingService: https://github.com/OpenTracksApp/OpenTracks/issues/703#issuecomment-816833733

commit 18cc3a3d (2021-04-09): Intervals are computated through TrackStatisticsUpdater. Fixes #703.
commit f45a1903 (2021-04-09): Intervals are computated through TrackStatisticsUpdater. Fixes #703.
commit e13735cb (2021-04-16): Intervals are computated through TrackStatisticsUpdater. Fixes #703.
commit ff7b7ba3 (2021-04-18): Intervals are computated through TrackStatisticsUpdater. Fixes #703.
closed 2021-06-15 by rgmf