OpenTracksApp/OpenTracks #710: Allow App Backup

sebastianha, 2021-04-14:
I just played around with seedvault as backup and it would not make a backup of open tracks and it shows: "App doesn't allow backup".

Is this intentional? I assume that it is not properly configured, see here: https://github.com/seedvault-app/seedvault/issues/81

dennisguse, 2021-04-14:
Yes, it is disabled on purpose: https://github.com/OpenTracksApp/OpenTracks/issues/346

If you want to enable this, you can build OpenTracks yourself.

closed 2021-04-14 by dennisguse

sebastianha, 2021-04-14:
Perhaps it would make sense to enable this for fdroid? Because this is also used by seedvault which is the open source backup solution built in into lineage os.

dennisguse, 2021-04-14:
Sadly that would be an anti-feature (IMHO) as this allowBackup will (also) enable backups to Google on phones with Google Play. That sucks hard and therefore it is disabled in OpenTracks.
When Google changes this, we can enable it. Some more infos are in the Android documentation: 

https://developer.android.com/guide/topics/data/autobackup

sebastianha, 2021-04-14:
Ok, thanks for the explanation!
That's really sad :(

sebastianha, 2021-04-17:
Sorry for still annoy you with this:

Wouldn't it be an option to allow backups but only when they are client side encrypted? Then also google should not have access even it gets backed up accidently.

https://github.com/seedvault-app/seedvault/wiki/FAQ#why-do-some-apps-not-allow-to-get-backed-up
https://developer.android.com/guide/topics/data/autobackup#define-device-conditions

dennisguse, 2021-04-19:
@sebastianha a PR is welcome; if you want to work on this, just re-open this issue.
Back then I just chose the safest/simplest option.
I would just like to be sure that enabling backups is not limiting privacy.

sebastianha, 2021-04-22:
I am sorry but my last app development was long time ago. Perhaps at some time in the future I will again get into it. I would be more than happy to create a PR.

dennisguse, 2021-04-22:
@sebastianha in this case it is more digging through the documentation.
The rest is just configuration.

mamoit, 2023-03-09:
@sebastianha
FYI: backups are now a thing :slightly_smiling_face:

sebastianha, 2023-03-09:
I just read the changelog, THANKS!