OpenTracksApp/OpenTracks #711: No voice announcements (microG)

mdkcore0, 2021-04-15:
**Describe the bug**
This is happening on a clean install of a ROM without gapps (using microG). This was working on the stock image (Pixel 2 XL), but after moving to this gapps-free installation can't make voice announcements to work.

NOTE: Have RHVoice installed and it's working on OsmAnd~ and FitoTrack (also have tested eSpeak)

**To Reproduce**
1. Go to Settings
2. Enable voice (Set voice frequency for 1 min, for example)
3. Start recording
4. See error

The only info I found is this from logcat:
`04-15 16:18:31.767 32625 32675 I AnnouncementPeriodicTask: TTS not ready.`

**Technical information**
 - Device: Google Pixel 2 XL
 - OS: ProtonAOSP 11.3.3 (sans gapps, but on microG)
 - Version v3.16.0

dennisguse, 2021-04-15:
Disclaimers: 
* this code is ancient ;)
* I have no TTS setup (so not testing on my side)

I checked the log and the TTS system is fails to init immediately (TextToSpeech.ERROR).
Don't know why it happens twice.
```
04-15 16:41:34.681  2427  2427 D AnnouncementPeriodicTask: Start
04-15 16:41:34.682  2427  2427 I AnnouncementPeriodicTask: TextToSpeech initialized with status -1
04-15 16:41:34.682  2427  2427 W TextToSpeech: shutdown failed: not bound to TTS engine
04-15 16:41:34.682  2427  2427 D AnnouncementPeriodicTask: Start
04-15 16:41:34.683  2427  2427 I AnnouncementPeriodicTask: TextToSpeech initialized with status -1
```

Finding:  your Android reports not TTS available for OpenTracks.

I noticed that you use Android11 (aka SDK30) and OpenTracks targets for SDK30.
According to the documentation, we may need to extend the AndroidManifest.xml:, so that Android11 also allows OpenTracks to access the TTS.
```   ...
<queries>
  <intent>
      <action android:name="android.intent.action.TTS_SERVICE" />
  </intent>
 </queries>
```

https://developer.android.com/reference/android/speech/tts/TextToSpeech

@mdkcore0 Can you compile and test yourself?

mdkcore0, 2021-04-15:
> @mdkcore0 Can you compile and test yourself?

sure, will try later tonight, thanks!

mdkcore0, 2021-04-16:
> ```
> <queries>
>   <intent>
>       <action android:name="android.intent.action.TTS_SERVICE" />
>   </intent>
>  </queries>
> ```
> 
> https://developer.android.com/reference/android/speech/tts/TextToSpeech

it works, now I have voice announcements again!!
```
04-16 01:15:06.870 14415 14415 I TrackRecordingServiceConnection: Connected to the service.
04-16 01:15:06.897  3717  3732 W rhvoice.androi: Reducing the number of considered missed Gc histogram windows from 502 to 100
04-16 01:15:06.919 14415 14415 I TextToSpeech: Connected to ComponentInfo{com.github.olga_yakovleva.rhvoice.android/com.github.olga_yakovleva.rhvoice.android.RHVoiceService}
04-16 01:15:06.923 14415 14594 I TextToSpeech: Set up connection to ComponentInfo{com.github.olga_yakovleva.rhvoice.android/com.github.olga_yakovleva.rhvoice.android.RHVoiceService}
04-16 01:15:06.932 14415 14415 I AnnouncementPeriodicTask: TextToSpeech initialized with status 0
```

can I open a PR with this fix?

dennisguse, 2021-04-16:
A PR  would be great!

mdkcore0, 2021-04-16:
done, opened #713, all testes passed :)

commit 3c5a1be1 (2021-04-17): Fix #711: declare TextToSpeech.Engine.INTENT_ACTION_TTS_SERVICE on queries element

This enables the use of TextToSpeech engine on Android 11.
Ref.: https://developer.android.com/reference/android/speech/tts/TextToSpeech
closed 2021-04-17 by dennisguse