OpenTracksApp/OpenTracks #718: ALTERNATIVE CRYPTOCURRENCIES FOR DONATIONS

G-i-o, 2021-04-17:
Thank You for providing this invaluable app for those who care about their privacy and opensource projects.

Thank You also for accepting (a) cryptocurrency for donations as it makes it much easier for some and also gives a choice to those who want to support alternative, somewhat more ethical, opensource and decentralized finance.

With this, I'd like to encourage you to add at least one or more cryptocurrencies. The reasons being the environmental issues and the relatively high costs annexed to BTC mining and, generally to all or most PoW coins. In particular, with BTC, there are elevated incentives to produce electronic devices (Asic miners) which are most times used for relatively short periods and then discarded with currently narrow changes of being used for other purposes.
As you may know, electronic devices contain highly polluting minerals and other materials that are very difficult to recycle and generally speaking are most often released into the environment one way or another. All this, setting aside the (real) mining of minerals which comes with associated abuse of human rights and further negative environmental impacts.

Also, given that we have mentioned privacy, there's also the issue of BTC transactions being fully public.

If I may, I'd like to suggest to add a (or more) PoS coin, such as Cardano, Polkadot, Ethereum (which will turn a PoS soon) and many more to mitigate the above mentioned environmental issues and a data-protection coin such as PIVX if you want also to offer the option of additional privacy to your donors. This can easily be achieved by either having a multicoin wallet, such as Coinomi, Flits, AtomicWallet etc., or using a payment gateway, which usually offers FIAT conversion and other tools for merchants and organizations and have minimal charges (Usually around 0.5-1%).

Thank You for reading and considering and keep up for the good work. 🙏

dennisguse, 2021-05-05:
Which one?

G-i-o, 2021-05-15:
Hello Dennis.

If you asked me, I'd add PIVX, Cardano, Polcadot and Ethereum 2 when is out (should happen around next year).

They are all environmentally sound and PIVX adds the privacy too (but it's not as well known as the others). So, I would add all the above or at least one among them.

Thank YOU!

dennisguse, 2021-10-16:
I don't see this happening; I got about 150USD in Bitcoin and actually costs me more time to get it transferred than it is worth.
It will take some effort to accept different coins and that is just too distracting.

@rgmf @pstorch If you see this differently, feel free to have fun in this domain.

closed 2021-10-16 by dennisguse

pstorch, 2021-10-16:
Actually I'm not a fan of cryptocurrency. From an environmental point of view.
Recently I've removed the Bitcoin donation link from the OSMDashboard fdroid metadata.

G-i-o, 2021-10-16:
> Mobile for iOS or Android.

Hi. I respect your decisions but in my view it's a
pity that you guys throw a bunch of inaccurate
info just before closing the discussion instead of
asking first what solutions could be there for the
half-non existent issues you mentioned.

1) Actually, by using a payment gateway, like
Coinpaymens just to mention one, you don't need to
do anything. You would receive your FIAT money in
your bank account with no hassle at all. Not only
that, but you would be paying far lower fees than
Paypal, for instance.

2) Nowadays, only fully unaware people would be so
silly to use BTC as currency, which obviously is
not (tx time and cost for each tx make it
impossible to take that role). There are a bunch
of other currencies that can be used but, again,
by using a payment gateway you wouldn't need to
worry about that.

3) I am also a fanatic about the environment. I
don't fly, don't eat meat etc. However, while it's
true that BTC is not good for the environment,
this is often more propaganda than an actual
issue. Partly for what mentioned above (e.g. BTC
cannot be used as currency) and partly because,
placing a bit of perspective, we can also see that
Visa and Paypal are also extremely bad for the
environment...yet something is going to tell me
that those won't be avoided because of it. 😛️
Again, there are alternative cryptocurrencies
(from Cardano, academically developed, Polkadot,
soon Ethereum 2,  and practically all those based
on a Proof of Stake algorithm ones....which BTC is
not). For example, the PIVX community and devs
once calculated that to run their whole
blockchain, it would suffice a single wind
turbine.

There are also few other environmental as well as
ethical advantages in chosing crytocurrencies over
FIAT money but I wont' bother you longer.

You obviously have already made your decision but
it's just a pity that it's based on incorrect
conclusions.

Anyhow, I appreciate all your work and hope long
prosperous live to the project.

G-i-o, 2021-10-17:
> I prefer if people just contribute to OpenTracks
> (either directly aka coding, proposing features,
> or finding bugs; or indirectly aka tell other
> people that it is a great tool) - it is way
> easier and also feels less like actual work (aka
> day job).
> It is great to keep to get some donations/ sales
> (PlayStore) to get compensated for the costs
> (devices like sensors, domain etc).
> @G-i-o
> PS/ Using a payment provider is actually the
> opposite of how I would use cryptocurrencies as
> that is also just another bank.
> The bitcoin address provided in the OpenTracks
> README was created on one of my machines
> locally.
> It is not the fastest device (but a really
> awesome one), and it took 8 days to download the
> blockchain and verify it.
> If I now want to send the bitcoins anywhere, I
> need to download the additional changes - might
> take another 2 days.
> And then I have the problem that I cannot use
> bitcoins for anything except exchanging it to
> FIAT money (at least I don't know any place
> where I could buy something useful with
> bitcoins).
> So, it feels like getting donations in a weird
> currency (requiring quite some effort to do it
> properly) and then getting it exchanged into
> EUROs to actually do something with it.
> It is probably easier (and also kind of privacy
> friendly) to send donations as postal stamps ;)

Apologies for my long replies but there are a
number of inaccuracies that I feel to address.

The bullets below follow roughly the order of the
critiques made above.

∙ One contribution type does not deny the other.
One can make contributions in multiple ways. Just
because I want to make a donation does not mean
that I would stop also reporting bugs etc.

∙ I agree, the  best would be to use
crytocurrencies direclty without intermediary.
However, it was yourself who mentioned that it was
time wasting etc for you. That's the only reason
why I mentioned the gateway....as ONE of the
available options. Also, services like Coinpayment
may even be used simply as managers, there no need
to convert if one does not want to. Beside, even
economically would make sense to hold the cryptos
if you don't need to spend them straight away.
By any side ones tackle it, you can always find
both ethical and practical advantages compared to
use alterative FIAT methods (including Liberapay
etc.....which by my understanding charge at least
4%.).

∙ As mentioned before, for me BTC it's not a
viable cryptocurrency at all. So, sorry for your
troubles with it but I am not suggesting to use
BTC at all.

∙ You can use cryptos for buying stuff or services
directly. It is incorrect to say that one cannot
use them but to exchange them with FIAT. For
examples see Bitrefill, where you can buy mobile
top ups, supermarkets (and many other shops from
all over the world) giftcards, travel tickets and
so forth. On Cryptwerk you can also find a list of
many shops and services for cryptos. Finally, some
providers, mostly VPN, servers/storage/cloud
providers etc. accept cryptos for payments.

∙ Why "weird". Cardano is accademically developed.
Ethereum is currrently bigger than BTC. Polkadot
is developed by one of the founder of Ethereum,
PIVX is 100% DAO community based etc. all these
are on the very top of the markets in terms of
daily transactions and value.
 
∙ According to how one goes, the bare minimum
needed would be to publish the coin receiving
address. You DO NOT need to download any
blockchain. There are a number of opensource
multi-coin light clients, like Guarda, Coinomi
etc. both on desktop and mobile that are extremely
light in resource and can hold and transact
hundred of coins.

Hopefully I'm not missing any point but I'm happy
to provide further info if needed. 

My feeling though, is that you have already made
your decision that won't change regardless of the
advancement that the crypto world has made in the
last few years.

🙏️

dennisguse, 2024-02-14:
@serrq if you really want: just create a new wallet and send it to me by email.