OpenTracksApp/OpenTracks #719: MemoryLeak as TrackRecordingListeners are not unregistered

dennisguse, 2021-04-18:
**Describe the bug**
When we refactored to use listener to communicate the TrackRecordingService, we are only registering but __we never unregister__.
Therefore, we (a) keep objects in memory that are not used anymore until TrackRecordingService shuts down and (b) we inform objects about changes that are already finished.

**To Reproduce**
I saw this issue while executing tests that resulted in NPE as the TrackRecordingService was re-used for a recording.
The NPE was triggered in the AnnouncementPeriodTask as it was already shutdown, but received an update from TrackRecordingService.

I got this issue due to 772dad48372c4fba9e4de2afc89fb7ab42ee712c
(not the actual reason, but not correct solution anyhow).

**Technical information**
 - Version v3.17.1

rgmf, 2021-04-18:
@dennisguse Are you working on it?

dennisguse, 2021-04-18:
@rgmf Not yet; took me some time too figure it out.
If you have some time go for it.
#720 contains a fix for the failing tests (observed this on locally) by refactoring the AnnouncementPeriodicTask.

dennisguse, 2021-04-20:
@rgmf I would have time to work today.
Did you do something here?

My plan:
* implement unregister from components
* Use WeakReference to keep reference in ServiceStatus
* Adding a RuntimeException in the service's onDestroy if listener is not empty (for testing)

dennisguse, 2021-04-20:
And the problem is actually described on Wikipedia: https://en.wikipedia.org/wiki/Lapsed_listener_problem

rgmf, 2021-04-20:
> @rgmf I would have time to work today.
> Did you do something here?
> 
> My plan:
> 
> * implement unregister from components
> * Use WeakReference to keep reference in ServiceStatus
> * Adding a RuntimeException in the service's onDestroy if listener is not empty (for testing)

No, I don't have time until weekend.

The plan sound great. If you cannot do it I could do it next weekend.

rgmf, 2021-04-20:
> And the problem is actually described on Wikipedia: https://en.wikipedia.org/wiki/Lapsed_listener_problem

Wow!! What a problem I introduced here :see_no_evil:

dennisguse, 2021-04-22:
I did some digging and the situation is not as bad as it could be.

Findings:
* nobody is unregistering from ServiceStatus
* ... but when TrackRecordingService is destroyed all reference are cleared
* we do not leak too much memory as only the Activity+TrackDataHub classes are leaking
  (not any large objects like UI)
* TrackDataHub is registering for recordingTrackId, but I don't understand why. IMHO should the trackId be provided by the activity; also it is not useful if a recorded track is shown.
* No crashes due to getting the change notification in the wrong state like a destroyed activity

To reproduce:
* start a recording
* move from TrackRecordingActivity ot TrackListActivity and back

Just printing the ServiceStatus operations that change listener:
```
2021-04-22 20:05:54.237 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.content.TrackDataHub@5a652c0]
2021-04-22 20:05:54.255 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus CHANGE: [de.dennisguse.opentracks.content.TrackDataHub@5a652c0]
2021-04-22 20:05:54.339 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.content.TrackDataHub@5a652c0, de.dennisguse.opentracks.TrackRecordingActivity@b4e6b62]
2021-04-22 20:05:54.369 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.content.TrackDataHub@5a652c0, de.dennisguse.opentracks.TrackRecordingActivity@b4e6b62, de.dennisguse.opentracks.TrackListActivity@2537704]
2021-04-22 20:05:57.461 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus CHANGE: [de.dennisguse.opentracks.content.TrackDataHub@5a652c0, de.dennisguse.opentracks.TrackRecordingActivity@b4e6b62, de.dennisguse.opentracks.TrackListActivity@2537704]
2021-04-22 20:05:57.617 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus stopping: [de.dennisguse.opentracks.content.TrackDataHub@5a652c0, de.dennisguse.opentracks.TrackRecordingActivity@b4e6b62, de.dennisguse.opentracks.TrackListActivity@2537704]
2021-04-22 20:06:00.835 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.TrackListActivity@2537704]
2021-04-22 20:06:00.866 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.TrackListActivity@2537704, de.dennisguse.opentracks.content.TrackDataHub@aa8a9a5]
2021-04-22 20:06:00.884 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus CHANGE: [de.dennisguse.opentracks.TrackListActivity@2537704, de.dennisguse.opentracks.content.TrackDataHub@aa8a9a5]
2021-04-22 20:06:00.944 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.TrackListActivity@2537704, de.dennisguse.opentracks.content.TrackDataHub@aa8a9a5, de.dennisguse.opentracks.TrackRecordingActivity@6eeb7db]
2021-04-22 20:06:04.315 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.TrackListActivity@2537704, de.dennisguse.opentracks.content.TrackDataHub@aa8a9a5, de.dennisguse.opentracks.TrackRecordingActivity@6eeb7db, de.dennisguse.opentracks.TrackRecordingActivity@ab25e63]
2021-04-22 20:06:04.316 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.TrackListActivity@2537704, de.dennisguse.opentracks.content.TrackDataHub@aa8a9a5, de.dennisguse.opentracks.TrackRecordingActivity@6eeb7db, de.dennisguse.opentracks.TrackRecordingActivity@ab25e63, de.dennisguse.opentracks.content.TrackDataHub@f3e0317]
2021-04-22 20:06:05.986 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.TrackListActivity@2537704, de.dennisguse.opentracks.content.TrackDataHub@aa8a9a5, de.dennisguse.opentracks.TrackRecordingActivity@6eeb7db, de.dennisguse.opentracks.TrackRecordingActivity@ab25e63, de.dennisguse.opentracks.content.TrackDataHub@f3e0317, de.dennisguse.opentracks.TrackRecordingActivity@9d2ee18]
2021-04-22 20:06:05.987 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.TrackListActivity@2537704, de.dennisguse.opentracks.content.TrackDataHub@aa8a9a5, de.dennisguse.opentracks.TrackRecordingActivity@6eeb7db, de.dennisguse.opentracks.TrackRecordingActivity@ab25e63, de.dennisguse.opentracks.content.TrackDataHub@f3e0317, de.dennisguse.opentracks.TrackRecordingActivity@9d2ee18, de.dennisguse.opentracks.content.TrackDataHub@89d853c]
2021-04-22 20:06:07.491 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.TrackListActivity@2537704, de.dennisguse.opentracks.content.TrackDataHub@aa8a9a5, de.dennisguse.opentracks.TrackRecordingActivity@6eeb7db, de.dennisguse.opentracks.TrackRecordingActivity@ab25e63, de.dennisguse.opentracks.content.TrackDataHub@f3e0317, de.dennisguse.opentracks.TrackRecordingActivity@9d2ee18, de.dennisguse.opentracks.content.TrackDataHub@89d853c, de.dennisguse.opentracks.TrackRecordingActivity@5cbdfce]
2021-04-22 20:06:07.491 21240-21240/de.dennisguse.opentracks.debug E/ServiceStatus adding: [de.dennisguse.opentracks.TrackListActivity@2537704, de.dennisguse.opentracks.content.TrackDataHub@aa8a9a5, de.dennisguse.opentracks.TrackRecordingActivity@6eeb7db, de.dennisguse.opentracks.TrackRecordingActivity@ab25e63, de.dennisguse.opentracks.content.TrackDataHub@f3e0317, de.dennisguse.opentracks.TrackRecordingActivity@9d2ee18, de.dennisguse.opentracks.content.TrackDataHub@89d853c, de.dennisguse.opentracks.TrackRecordingActivity@5cbdfce, de [... cut ...] nisguse.opentracks.content.TrackDataHub@f3e0317, de.dennisguse.opentracks.TrackRecordingActivity@9d2ee18, de.dennisguse.opentracks.content.TrackDataHub@89d853c, de.dennisguse.opentracks.TrackRecordingActivity@5cbdfce, de.dennisguse.opentracks.content.TrackDataHub@51c1812, de.dennisguse.opentracks.TrackRecordingActivity@8853e07, de.dennisguse.opentracks.content.TrackDataHub@62a9b7b, de.dennisguse.opentracks.TrackRecordingActivity@d9f7b6c, de.dennisguse.opentracks.content.TrackDataHub@1e6d450]
```

dennisguse, 2021-04-22:
Now the solutions:

1. implement unregister on TrackRecordingServiceStatus via TrackRecordingService
  + easy to do
  - TrackRecordingService might not be available when we want to unregister (we are connecting to it via a ServiceConnection); and it might be dead?
   (is this case a problem?)
2. Distribute status changes via LocalBroadcast (initial status via TrackRecordingService as it is now happening by subscribing)
+ subscribe/unsubscribe happens on an object that is always available
+ components are independent of the service
- communication via intent might be more complex

3. Go back to sharedpreferences
  + was working
  - results in I/O
  - adds complexity to preferences
  (supports automatic resume after crash; we removed this functionality)

Also:
* [ ] TrackDataHub does not register at TrackRecordingService but gets all data from the activity

(Somehow I am in favor of 2.)

rgmf, 2021-04-22:
Option 2) looks good, but isn't LocalBroadcast deprecated in Android?

dennisguse, 2021-04-23:
@rgmf you are right.
Then I guess we may rather use:

4. LiveData

closed 2021-05-03 by dennisguse

dennisguse, 2021-05-09:
Released in v3.18.0

PR #731 (2021-04-27): Some improvements for #719
Does not fix #719.

TrackDataHub does not need to register for TrackRecordingService status changes as this can be handled by TrackRecordingActivity alone.

* reduces impact of memory leak as TrackDataHub keeps references to fragments; now only the activities are referenced by TrackRecordingServices and those at least remove all there references.
* (so far) TrackDataHub started TrackRecordingSevice due to binding to the service (even if only showing already recorded tracks)
commit e0d2636d (2021-04-29): TrackRecordingService: provide status updates via LiveData.

Fixes #719
commit 241202f5 (2021-04-29): TrackRecordingService: provide status updates via LiveData.

Fixes #719
commit f0bbf61b (2021-04-29): TrackRecordingService: provide status updates via LiveData.

Fixes #719