OpenTracksApp/OpenTracks #722: newest bugs after update

RedRainHag, 2021-04-20:


dennisguse, 2021-04-20:
Yes, the v3.17-series has some issue that we missed before release.

>     * can't export my runs anymore. I get the message that their was an error for all the exports...

Don't know what the issue is; can share some more infos (best case logcat).

> 
>     * I was using voice announcements for every 5km but it kept on announcing without pause. So I had to turn off the announcements. 

Tracked as #721; it is fixed and released as v3.17.3.
As of today, this is not yet available via F-Droid.

> 
>     * downgrade to older version viva Fdroid always resulted in an error that the installation was aborted.

Do you have some error message for this?
Because it should be possible; not database changes were included.
So, it should just work...

>     * app crashed several times after the update to 3.17,

Best guess is #719.
Also partly affected #721.
We are working on it.

> 
> 
> Thank you for your good work!
Thanks!

RedRainHag, 2021-04-21:
Thanks for the reply! 

> Do you have some error message for this?

To the downgrade I can't tell you much, I just get the message from F-Droid that the installation was aborted due to an unknown error and thats it.. 

> Don't know what the issue is; can share some more infos (best case logcat).
where can I get to the logcat? Not sure if that one still exists since I deinstalled and reinstalled the program after that...

dennisguse, 2021-04-21:
@RedRainHag
regarding logcat.
You need to enable developer mode on your device: https://developer.android.com/studio/debug/dev-options
And then you can use AndroidStudio to get the logcat: https://developer.android.com/studio/debug/am-logcat

Fregf, 2021-05-02:
> > * I was using voice announcements for every 5km but it kept on announcing without pause. So I had to turn off the announcements. 
> > ```
> 
> Tracked as #721; it is fixed and released as ~v3.17.3~ v3.17.2.
> As of today, this is not yet available via F-Droid.

This bug is still present for me in 3.17.3 currently in f-droid: I have it set to give announcements every 1 km, but it was constantly announcing.

dennisguse, 2021-05-02:
@Fregf by just a quick look: yes that might be broken as well due to the distance refactoring.
I will check this during the week; it will take some time as we have very little test coverage for the announcements.

closed 2021-05-05 by dennisguse

dennisguse, 2021-05-09:
Released in v3.18.0

PR #743 (2021-05-05): Bugfix: voice announcements need to use KM/Miles (instead of m/feet).
Fixes #722.

While introducing Distance, I used m/feet instead of KM/Miles to trigger the voice announcements based upon distance.

Reason: lack of test coverage.
commit b395c198 (2021-05-05): Bugfix: voice announcements need to use KM/Miles (instead of m/feet).

Fixes #722.