OpenTracksApp/OpenTracks #730: Android 11 doesn't record trips

rafaelmuynarsk, 2021-04-23:
**Describe the bug**
After my phone updated to Android 11 it doesn't receive this app is not updating location when the screen is off. I know a little of Android development too and tried to fix it myself. However, I didn't manage to fix it, on Google's official documentation they say that on Android 11 we need to request the permission `ACCESS_BACKGROUND_LOCATION` separately in order to access the location in background. So I tried changing the manifest and requesting it after the app opens. Even though the permission was granted, the app is still not recording my trip when the screen is off.

Google documentation telling about Android 11:  https://developer.android.com/about/versions/oreo/background-location-limits
StackOverflow answer that seems related to this bug: https://stackoverflow.com/a/65401607/14739472

**To Reproduce**
1- Use a phone with Android 11
2- Touch the button to start recording
3- Turn off the screen and make a trip
4- That's it, there won't be any trip recorded following these steps

If applicable:
To reproduce it on Android Studio I tried the following:
[image]

1- I went to the class LocationHandler
2- Inside the method OnLocationChanged I wrote `System.out.println("stringTest: " + location.getLongitude());`
3- Compiled the app and started watching its behavior. If I use a phone that's Android 10 or bellow... I can see the location being updated while recording even when my screen is off. When I use a phone with Android 11, it stops updating the location as soon as I turn off my screen... And if I turn on the screen again, it goes back to updating the location (it doesn't matter if OpenTracks is open or not, the trigger here is the screen only)

Additional information:

In theory, the only thing that is needed to solve this problem is adding the following permission on the Manifest file:

`<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />`
    
And then add this permission as a dangerous permission on some part of the Activity:

```
if (ContextCompat.checkSelfPermission(context, Manifest.permission.ACCESS_BACKGROUND_LOCATION) != PackageManager.PERMISSION_GRANTED) {
    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.Q) {
        ActivityCompat.requestPermissions(activity, new String[]{Manifest.permission.ACCESS_BACKGROUND_LOCATION}, 2); }
}
```

Well, that's what I've tried. But I failed to fix this problem. I'm not even sure if getting the location while the screen is off is still possible on Android 11. Any ideas on what to do to fix this issue?

**Technical information**
 - Device: Galaxy S10e
 - OS: Android 11
 - Git Commit: v3.17.2

rgmf, 2021-04-23:
I don't have a phone with Android 11 but I've tried to reproduce the issue on AVD and I couldn't do it. I don't know if we can trust on AVD for these things.

Have you disabled battery optimization for OpenTracks on your phone? Other users have had problems with that on other Android's versions:

Do you think it could be related to these?

https://github.com/OpenTracksApp/OpenTracks/issues/177

https://github.com/OpenTracksApp/OpenTracks/issues/596

rafaelmuynarsk, 2021-04-23:
I don't think they are related... I'm aware of the battery optimization settings and that I need to "Allow all the time" the location permissions:

[image]
[image]

It could be something specific with Samsung too... I don't know. But I'm sure that it stopped working after my phone updated from Android 10 to Android 11. I've tested with a different phone using Android 10 and it's working. And my Samsung S10e worked very well when it was Android 10. Now that it is Android 11 it stopped working.

rgmf, 2021-04-23:
So it seems other issue like you point out.

I cannot to figure out what the issue is because I cannot reproduce the issue from an AVD, sorry. I hope someone here can help you ;)

PS/ It's the first time I see "Allow all the time" for OpenTracks.

rafaelmuynarsk, 2021-04-23:
Thanks

The "Allow all the time" appears after you insert the line `  <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />` on the manifest file. By default it doesn't have it... But I tried that to see if it would solve the issue.

dennisguse, 2021-04-23:
It should not be ACCESS_BACKGROUND_LOCATION as we are starting a foreground service and therefore only need the foreground permission.
(Note: we should add a comment in the AndroidManifest.xml)

So, I guess it is some kind of powersave setting.
Is the battery saver enabled?
(or what is the battery icon in the status bar on the first screenshot)

@rafaelmuynarsk Nice report by the way!

rafaelmuynarsk, 2021-04-23:
@dennisguse Thanks! You're right... The powersave setting was enabled and I didn't realize it. After turning it off the location continues updating after turning off my screen.

My mistake, the issue is solved now. :)

dennisguse, 2021-04-23:
Only partly as the powersaving mode is an actual issue.
So it might be great, to check if powersaving is enabled (and therefore GPS is not used) and warn the user.

i guess #482 is related.

closed 2021-05-16 by dennisguse