OpenTracksApp/OpenTracks #732: Recording problem

Papipio, 2021-04-25:
Bonjour,

I like very much OpenTracks with offline OSM Dashboard maps, but :

1 - It is impossible to record correctly a trace, some portions are often missing, even recording is OK with AllTrails running simultaneously (I prefer OpenTracks !).
I checked all parameters, they are the same for both apps (no restriction for battery usage).
What can I do for having OpenTracks recording correctly ?

2 - It would be very interesting to be able to move the map, and with a new button to be able to reset the map position with the current position in the center of the display.

3 - For vocal notifications every km:
- It would be nice to add a small delay between the different sentences for being easier to understand,
- All sentences must be in French.

4 - For being able to display maps, I use MAPS.ME with Open Street Maps:  is it possible to use with OpenTracks the map files already stored for MAPS.ME ?

My smartphone is Xiaomi Mi 9 Lite.

Pierre

pstorch, 2021-04-25:
1. Maybe related to bad GPS Signal. Check setting GPS Accuracy. If accuracy is worse, then OT doesn't record track points.
2. You can move the map already and if a new track point of OT is received, the map centers on that position. There is no way for past tracks. OSMDashboard doesn't have GPS permission itself.
3. maybe @dennisguse or @rgmf have some input here
4. There are several different map formats out there. The OSMDashboard uses the maps forge library. You can find several map sources in the README. Unfortunately maps.me uses a different format.

rgmf, 2021-04-28:
About 3) A delay between the different sentences would be good. About French sentences: @Papipio if you know how to translate them and you are French would be nice if you could add them.

dennisguse, 2021-04-28:
Regarding 3) I guess the issue with the delay is that we actually do not have sentences there - at least they are not separated by a "."; and it might be that this differs by translation.
French and English do not have punctuation character in the current translations.

(Translations are already available in French)

closed 2021-05-16 by dennisguse