OpenTracksApp/OpenTracks #735: Implement deletion track according to Android lifecycle

rgmf, 2021-04-27:
**Is your feature request related to a problem? Please describe.**
Follow-up #734.

Read comment: https://github.com/OpenTracksApp/OpenTracks/pull/734#issuecomment-826386091

Basically we are using a Thread that will update UI if Activity is started/resumed. We currently have a collateral issue: when user tap back button while track is deleting then the ActivityList show the track already delete in background by the thread. Usually this is difficult because deletion is fast but it can be occur.

Sum up: we need another approach for this.

**Describe the solution you'd like**
Using an approach that match properly with Android Lifecycle.

rgmf, 2021-04-28:
I think the best solution should be a service so the TrackListActivity can bind to it and receive delete result by the service. But I have some doubts:

TrackListActivity will has to try to bind to the service. Can this be a problem? I mean, every time TrackListActivity is resumed or started it'll has to check if delete service is running to bind to it.

@dennisguse do you think that can be a problem? Am I missing something?

In that case some classes have to be edited:
- TrackListActivity.
- TrackRecordedActivity.
- AbstractListActivity.
- TrackDeleteActivity.

dennisguse, 2021-04-28:
Don't know - but I guess that won't be a (big) problem.
One data point from TrackDataHub: it was always trying to bind the service until recently.

Can we somehow get rid of TrackDeleteActivity by moving this to a separate service?
And I guess only TrackListActivity would need to check the progress - assuming that TrackRecordedActivity triggers deletion and then finishes. If TrackListActiivty then prevents opening Tracks that are in deletion, we could avoid this.

rgmf, 2021-04-28:
> Can we somehow get rid of TrackDeleteActivity by moving this to a separate service?

Yes, that's the idea.

dennisguse, 2021-04-29:
@rgmf Forgot: i am working on #736 (memoryLeak/liveData - hopefully done until the weekend) and this might create some conflicts with this issue.

rgmf, 2021-04-29:
@dennisguse okay, then I guess it'll be better to wait.

commit 95179d0d (2021-05-09): Track deletion is handled by a service that shows a progress bar on the notification bar. Fixes #735.
commit 06d4656e (2021-05-09): Track deletion is handled by a service that shows a progress bar on the notification bar. Fixes #735.
commit 3cb8a2a2 (2021-05-09): Track deletion is handled by a service that shows a progress bar on the notification bar. Fixes #735.
commit 5a3071a3 (2021-05-17): Track deletion is handled by a service that shows a progress bar on the notification bar. Fixes #735.
closed 2021-05-17 by rgmf