OpenTracksApp/OpenTracks #753: Exported / Imported moving time with pause.

rgmf, 2021-05-09:
**Describe the bug**
I recorded a long track and moving time not match after export/import it.

I figured it was because I paused the activity when we took a rest in the middle of the route to have a cafe. The difference between times looks like the time we took in that rest. But I've not been able to reproduce it.

The rest of stats are right.

[image]

dennisguse, 2021-05-09:
Also, the AVG SPEED is off.

Assumption: KML/KMZ

Possible things that might not work:
* KML exporter does not export PAUSE_MANUAL correctly
* KML importer does not import PAUSE_MANUAL correctly
* (unlikely) TrackStatisticsUpdater does something wrong

All these things do not have that good a test coverage.
We might start by making ExportImportTest reproducible, i.e., using fixed and useful timestamps for the test data instead of current.

PS: Nice gain!
Which I would have any mountains close by.

rgmf, 2021-05-09:
> Also, the AVG SPEED is off.
>
> Assumption: KML/KMZ
>
> Possible things that might not work:
>
>     * KML exporter does not export PAUSE_MANUAL correctly
>
>     * KML importer does not import PAUSE_MANUAL [...]

I was using GPX but I've tried exporting/importing in KML and it works well.

This issue only affects GPX files. Strange... I'll try to catch it.

> PS: Nice gain!
> Which I would have any mountains close by.

Here mountains are not very high but there are a lot of mountains :grinning:

dennisguse, 2021-05-09:
And I guess, what it is #597 that is tracking this missing feature.

closed 2021-05-21 by dennisguse