OpenTracksApp/OpenTracks #756: Elevation gain/loss not works

vsevjednom-cz, 2021-05-11:
Your code is amazing but:

There is elevation gain/loss value during measure but without values (- m). I don't know why.

**To Reproduce**
1. Tap to "start".
2. Look at the elevation gain/loss. There is still only "- m".

**Technical information**
 - Device: Honor KSA-LX9
 - OS: Android
 - Version 9.1.0

dennisguse, 2021-05-11:
I guess your telephone does not has a barometric sensor to measure pressure.
Using GPS altitude data is not (yet) implemented: #619

vsevjednom-cz, 2021-05-12:
I understand. Is possible to add "Show/hide elevation gain and loss" to the settings?

dennisguse, 2021-05-13:
@vsevjednom-cz At the moment, you  can only hide the altitude/elevation completely (absolute, gain/loss).

vsevjednom-cz, 2021-05-13:
Yes, I know. But hide elevation gain/loss should be the solution for mobile phones without barometric sensor 🙃.

dennisguse, 2021-05-13:
Yes, but no :)
Getting gain/loss from GPS values (with smoothing and some magic) is the actual solution aka #619.
I don't wont to spend time on a workaround. 
And the UI code is kind of complicated (less clean than it should be) - I am just reworking some part of the internal data transfer to avoid complexity.
It is a process....

TL;DR: If you want this now, go ahead.

Sonegide, 2021-05-29:
Hi there,
I am having the same issue, with the elevation gain/loss not showing in the first tab.
The odd thing is that it is visible in the charts of the third and fourth tabs.
On older tracks, with the same phone, it was only showing the elevation gain, but not loss (while I always ended where I started).

dennisguse, 2021-05-29:
@sonegide absolute elevation from the GPS is shown in the charts.
But these values cannot be directly used to get the elevation gain/loss.
We need to do some filtering etc before this is actual useful.
The issue is that GPS elevation is very noisy...

Sonegide, 2021-05-29:
I don't see any absolute value in the chart (if I understand correctly, that you mean total value).
In case, I include a screenshot.

In any case I agree with the noise, therefore it may not be so relevant for me to get this total at this stage.
Anyway, thank you for this app, it's great  :)

[image]