OpenTracksApp/OpenTracks #757: Search is not working from TrackListActivity

dennisguse, 2021-05-11:
Since July 2020....

Search from TrackListActivity is not working as the search action is only triggered if the menu item is shown as an action.
However, only two actions can be present and aggregated stats + GPS are more important.
This would fix the issue, but then search is shown instead of GPS status.
```
    <item
        android:id="@+id/track_list_search"
        android:title="@string/menu_search"
        app:actionViewClass="androidx.appcompat.widget.SearchView"
        app:showAsAction="always" />
```

dennisguse, 2021-05-11:
I am not really sure.
I really like that the aggregated stats are shown directly (by the way nice work!) as well as the GPS status.
It seems as if the search is not used that often or otherwise people would have informed us.

I see two options:
* replace the Toolbar with something that looks similar but allows for more icons
  (probably kind of complicated, but we waste a lot of space there)
* implement a 2nd activity that handles the search and is started in search mode
  (basically TrackRecordingActivity without ControllerFragment + menu)

For now, I was thinking about hiding the search functionality (as a quick fix).

sebastianha, 2021-05-15:
> I didn't realise search didn't work. Also nobody have opened an issue. Maybe it's not the more used feature. Anyway I wouldn't remove it.
> 
> I think hidding it and think what to do later is a good idea.
> 
> About the solutiona, the second option looks easy and a good solution.

I just stumbled upon this problem so here is a search user!

dennisguse, 2021-05-15:
@sebastianha Cool!
If you have the time, you may test: https://github.com/OpenTracksApp/OpenTracks/pull/772

sebastianha, 2021-05-15:
Hi, I have currently no build environment available, sorry. But I will test as soon as the next release is ready. Or is there any auto dev build available?

commit 948d9b1a (2021-05-15): TrackListActivity: re-enable search.

Fixes #757.
PR #772 (2021-05-16): TrackListActivity: re-enable search.
Fixes #757.

Search can be used if on search start, the menu item is upgraded to an Action (i.e., `searchMenuItem.setShowAsAction(MenuItem.SHOW_AS_ACTION_ALWAYS);`).

Also simplify search setup.
commit 6b0ec72b (2021-05-16): TrackListActivity: re-enable search.

Fixes #757.
closed 2021-05-16 by dennisguse