OpenTracksApp/OpenTracks #769: Automatic upload GPX file to the FTP server

vsevjednom-cz, 2021-05-13:
**Describe the solution you'd like**
I need upload GPX file to my FTP server at the background after finishing a track (only at the wifi ideally).

**Describe alternatives you've considered**
I must export track to the external storage and upload file to FTP server manually. The reason is simply: I created a website feature with OpenStreetMap and OpenLayers.
This is solution for FTP upload at the backgroud maybe: https://github.com/gotev/android-upload-service
But I'm affraid to make such a major change to the code. So I can help you but I don't know how.
I readed [this](https://github.com/OpenTracksApp/OpenTracks/issues/90) - foldersync works as I want. But setting this app is heavy. And I don't want create FTP account for all of my friends.

So is this plugin still planned? You wrote seomething about it [here](https://github.com/OpenTracksApp/OpenTracks/issues/90#issuecomment-561493507). I readed that you don't want use `android.permission.INTERNET` - but why in this case?

dennisguse, 2021-05-13:
_Disclaimer:_ `android.permission.INTERNET` is out of scope due to privacy/security reasons: OpenTracks cannot send data anywhere; this is completely at the user's liberty using the share feature or something like that.

You want an automatic upload of __all__ recorded tracks when connecting to a Wifi, right?
You may use any other app for this as OpenTracks can export a track after finishing recording to the external storage.
Then you may use something like SyncThing, OwnCloud/NextCloud etc.

Or people can use the share feature to send recordings manually, e.g., by mail to your server.

PS: Why FTP? Wouldn't be HTTP-based protocols be more robust?
And how do you handle authentication?
I presume you don't want to have anonymous uploads, right?

vsevjednom-cz, 2021-05-13:
You're right. It needs authentication. I have to do it for myself.

closed 2021-05-13 by vsevjednom-cz