OpenTracksApp/OpenTracks #776: Show warning if powersave mode is/ get's enabled

dennisguse, 2021-05-16:
Enabling powersaving mode usually disables the GPS and thus no track is recorded.
At least, we should show a warning to the user.

_Originally posted by @rafaelmuynarsk in https://github.com/OpenTracksApp/OpenTracks/issues/730#issuecomment-825883208_

ph818, 2021-09-10:
Would it be possible to show the user a warning if a lot of points are being discarded due to location inaccuracy? Inaccurate location signal and overzealous power saving can both lead to gaps in the recorded track and as a user it's hard to tell what's happening

dennisguse, 2021-09-10:
@ph818 good idea!
This was not a high priority so far though.

dennisguse, 2022-02-07:
Might actually be fixed by: https://github.com/OpenTracksApp/OpenTracks/pull/1114