OpenTracksApp/OpenTracks #778: Add tests for GPX and KML importer

dennisguse, 2021-05-16:
We have no test coverage for the GPX and KML importer (besides ExportImportTest), but we need this as the code needs to get refactored and it is hard to figure out what kind of data we support.

* [ ] GPXFileTrackImporter
* [ ] KMLFileTrackImporter
* [ ] (bonues) refactor both importer, so they don't inherit from AbstractFileTrackImporter, but use the same class for just creating the track (not being part of the actual XML parsing) 

Related to https://github.com/OpenTracksApp/OpenTracks/issues/775

balajiramanathan, 2021-05-31:
I would be happy to help if you can tell me what help is needed.  I can read and understand basic XML and can edit KML and GPX files to create test cases if needed.  I am not a Java coder but I am a programmer by training, so I can help a little with logic if needed though not with the actual coding.

dennisguse, 2021-06-01:
@balajiramanathan That would be awesome!
You could start with GPX as it is simpler (in terms of different options that could happen).
Some ideas:
* https://github.com/OpenTracksApp/OpenTracks/issues/774
* https://github.com/OpenTracksApp/OpenTracks/issues/683
* and multiple tracks per file

And the simpler the files the better - makes debugging and thus maintenance simpler:
```
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns:atom="http://www.w3.org/2005/Atom"
    xmlns:gx="http://www.google.com/kml/ext/2.2"
    xmlns="http://www.opengis.net/kml/2.2">
    <Document>
        <Placemark>
            <gx:MultiTrack>
                <altitudeMode>absolute</altitudeMode>
                <gx:interpolate>1</gx:interpolate>
                <gx:Track>
                    <when>2021-05-29T18:06:21.767Z</when>

                    <gx:coord />

                    <gx:coord>14.0 3.0 10.0</gx:coord>
                    <when>2021-05-29T18:06:22.042Z</when>

                    <when>2021-05-29T18:06:22.192Z</when>
                    <when>2021-05-29T18:06:22.318Z</when>
                    <when>2021-05-29T18:06:22.512Z</when>

                    <gx:coord>14.001 3.0 10.0</gx:coord>
                    <gx:coord>14.002 3.0 10.0</gx:coord>
                    <gx:coord />
                </gx:Track>
            </gx:MultiTrack>
        </Placemark>
    </Document>
</kml>
```

closed 2023-05-27 by dennisguse