OpenTracksApp/OpenTracks #779: All sql operations out of main thread

rgmf, 2021-05-17:
**Is your feature request related to a problem? Please describe.**
There are sql operations running on the main UI thread. In most cases there are not problems but they should be done in other threads.

**Describe the solution you'd like**
Every method in `ContentProviderUtils` return a `LiveData` and the rest of components can observe them to get the results. An example:

In ContentProviderUtils:
```java
    public MutableLiveData<Track> getTrack(@NonNull Track.Id trackId) {
        MutableLiveData<Track> liveData = new MutableLiveData<>();

        new Thread(() -> {
            try (Cursor cursor = getTrackCursor(TracksColumns._ID + "=?", new String[]{Long.toString(trackId.getId())}, null)) {
                if (cursor != null && cursor.moveToNext()) {
                    liveData.postValue(createTrack(cursor));
                } else {
                    liveData.postValue(null);
                }
            }
        }).start();

        return liveData;
    }
```

In StatisticsRecordedFragment:
```java
contentProviderUtils.getTrack(trackId).observe(this, track -> {
	if (track == null) {
	    Log.e(TAG, "track cannot be null");
	    getActivity().finish();
	    return;
	}
	
	sensorStatistics = contentProviderUtils.getSensorStats(trackId);
	
	boolean prefsChanged = this.track == null || (!this.track.getCategory().equals(track.getCategory()));
	this.track = track;
	if (prefsChanged) {
	    sharedPreferenceChangeListener.onSharedPreferenceChanged(sharedPreferences, getString(R.string.stats_rate_key));
	}
	
	loadTrackDescription(track);
	updateUI();
	updateSensorUI();
    });
```

dennisguse, 2021-05-18:
This also fixes:
```
D/StrictMode: StrictMode policy violation; ~duration=25 ms: android.os.strictmode.DiskReadViolation
        at android.os.StrictMode$AndroidBlockGuardPolicy.onReadFromDisk(StrictMode.java:1556)
        at android.database.sqlite.SQLiteConnection.applyBlockGuardPolicy(SQLiteConnection.java:1138)
        at android.database.sqlite.SQLiteConnection.executeForCursorWindow(SQLiteConnection.java:939)
        at android.database.sqlite.SQLiteSession.executeForCursorWindow(SQLiteSession.java:838)
        at android.database.sqlite.SQLiteQuery.fillWindow(SQLiteQuery.java:62)
        at android.database.sqlite.SQLiteCursor.fillWindow(SQLiteCursor.java:153)
        at android.database.sqlite.SQLiteCursor.getCount(SQLiteCursor.java:140)
        at android.content.ContentResolver.query(ContentResolver.java:963)
        at android.content.ContentResolver.query(ContentResolver.java:880)
        at android.content.ContentResolver.query(ContentResolver.java:836)
        at de.dennisguse.opentracks.content.provider.ContentProviderUtils.getTrackCursor(ContentProviderUtils.java:236)
        at de.dennisguse.opentracks.content.provider.ContentProviderUtils.getTrack(ContentProviderUtils.java:209)
        at de.dennisguse.opentracks.TrackRecordedActivity.onPrepareOptionsMenu(TrackRecordedActivity.java:192)
```

rgmf, 2021-06-15:
It won't be "fixed"... see https://github.com/OpenTracksApp/OpenTracks/pull/801

So I think it can be closed for now.

commit 04c40f65 (2021-05-29): Activity and Fragment classes do SQL operations out of UI thread. Fixes #779.
closed 2021-06-15 by rgmf