OpenTracksApp/OpenTracks #780: Continue Track does not work reliable

mega-stoffel, 2021-05-17:
**Describe the bug**
I usually start a recording at the beginning of the day when I leave the house and then I just stop this record and just "continue" it, whenever I'm back on the bike. Since the last version (got it via f-droid) it looks like this continue-function is not working reliably anymore.
When I hit "continue" I just put the phone in my pocket. And now I found out, that the app is sometimes/always (haven't found that out, yet) not recording anymore. When I look at the app, I just see the list of all recorded tracks - but it's not recording anymore. At first I just thought, I forgot it or didn't hit the correct button. But after I took care about it, I think, it's a misbehavior.
The Battery Savings option is not activated for this app.

What can I do to get more information to find and solve this bug?

**Technical information**
 - Device: Fairphone FP3
 - OS: Android 10
 - Version 3.18.0

abthediablo, 2021-05-18:
I can reproduce the issue with Samsung Note 9 and Android 10.

dennisguse, 2021-05-18:
Doesn't sound too good, but I did not fully understand what kind of problem you have.

1. Are you pausing a track (via pause button)?
2. ... or are you resuming a track (via context menu from a recorded track)?
3. Is the notification shown (should be there while recording)?
   (if it is not there OpenTracks is not recording / keeping GPS fix while paused)
4. What was the last working version (that you know of)?
5. Does it always happen or only sometimes?
6. And can you create a description how to get into this state?

@rgmf Any ideas?

rgmf, 2021-05-18:
@dennisguse I've not been able to reproduce it (resume crash). I've tried with two phones: Android 9 and Lineage 17.1. And also with AVD with Android 10.

We would need some logs from @mega-stoffel and/or @abthediablo... I don't know if it's feasible.

PS/ If issue would triggered every once in a while I'd bet the issue is the #696 one :thinking: I've stumbled upon this a couple of times for last two weeks.

mega-stoffel, 2021-05-18:
Hej!
1. No, I don't use pause
2. Yes, I usually stop the track with the stop button on the bottom right and quit the app afterwards.
3. Not sure, since I just put the phone in my pocket. But I will take care, now.
4. I get my updates via f-droid and would say, the previous version 3.17.3(?) was not behaving like this. I'm running 3.18 now.
5. It doesn't happen every time, sorry. Just today I tried to reproduce it, but today it worked while I was playing around with it.
6. I will try to find a way to reproduce this behaviour and let you know.

How could I get to the mentioned log files?

Thanks for taking care - and, by the way, thanks a lot for the app! :-)

abthediablo, 2021-05-19:
I had a little cycling trip today and using OpenTracks to record the distance of the current month. Beside I am using 4 other GPS-Apps, which are working fine, so I am sure it is not a problem concerning the android settings e.g. power saving.

So when I am start/continue recording the track then everything is fine. But at the following situations it seems that the app crashes / stops in the background:
- switching to anonther app
- try to add a waypoint
- turn of the display of my phone

So as long I will keep OpenTracks active on the screen everything is fine and it works perfect.

Well it seems that the problem came with the last update cause I ve already recorded 340km this month and did the last big trip (55km) at the 11th of mai. And well cause I am using f-droid I think the update came around 4/5 days ago.

dennisguse, 2021-05-19:
@abthediablo I can confirm this one; it is already fixed but not yet released.
I should be fixed by this commit: https://github.com/OpenTracksApp/OpenTracks/commit/18dd1de1f9c3355cc01c0efd5b34728f0b69f714
... but is not yet released.
It was a mistake in lifecycle handling that I introduced.

@mega-stoffel not yet sure if it is the same problem.

@rgmf Anything blocking us from releasing v3.18.1?

rgmf, 2021-05-19:
@dennisguse go ahead!

dennisguse, 2021-05-19:
@abthediablo @mega-stoffel 
If you have some time for testing, attached is v3.18.1 (PlayStore releases, can be installed parallel to F-Droid version).

[OpenTracksUpstream-releasePlayStore.zip](https://github.com/OpenTracksApp/OpenTracks/files/6510572/OpenTracksUpstream-releasePlayStore.zip)

abthediablo, 2021-05-19:
> @abthediablo @mega-stoffel
> If you have some time for testing, attached is v3.18.1 (PlayStore releases, can be installed parallel to F-Droid version).
> 
> [OpenTracksUpstream-releasePlayStore.zip](https://github.com/OpenTracksApp/OpenTracks/files/6510572/OpenTracksUpstream-releasePlayStore.zip)

Just installed it - will try it tomorrow on my way to work.

mega-stoffel, 2021-05-20:
yesterday I was on the road again and it happened again.
I started the app and rode to the train station. There I stopped it and travelled by train for about one hour. I started the app again in the other train station and is seemed to be ok so I put the phone in my pocket. After I rode through the city I looked at the phone and there was no recording anymore, just the list of my tracks. I didn't use the phone for anything else. I continued the track again and it worked now.
Some hours later I travelled back, stopped the recording in the train station and started it again (i.e. continued the existing track) in my home town. There it worked for about one kilometer and then the recording stopped. I didn't use the phone for anything, it was just in my pocket.
Here's a strava screenshot. On the left you can see, where it suddenly stopped to record. On the right there's some straight line to the place, where I opened the app again and found out, that it didn't record anymore, but only showed the list of the existing tracks.
[image]
Where can I find the logs of the recordings of yesterday?

dennisguse, 2021-05-20:
Hypothesis (a): What you describe sounds as if OpenTracks is crashing or being stopped by Android.
Just to verify this hypothesis: when OpenTracks is not running (because it got terminated) and it is then started, the splash screen is shown (dark background only the logo) briefly.
Thus, the start screen aka track list is shown again.
However, the destination should not be recorded (see below).

Hypothesis (b): GPS is turned off by the phone due to some power saving
OpenTracks stores a new TrackPoint (aka current location + timestamp etc) every time a new location is reported by the GPS.
As your destination is stored the service that actually does the recording is still running, but did not got any updates from the GPS in between (resulting in this straight line).
But the track list should not be shown in this case.
Actually only, if there is some malfunction in the UI.

PS: Both hypothesis are really ugly/nasty problems.

@mega-stoffel I am not sure how to proceed here.
Logs would help: https://developer.android.com/studio/debug/am-logcat
But don't share them publicly as they may contain sensitive information (filter by OpenTracks helps already).
Logs are deleted automatically (I guess): I don't know how long they stay on the device.

dennisguse, 2021-05-20:
@mega-stoffel For hypothesis (b), the diagram should look like this (straight line starting at 1:30:00). I don't know why this happened, but that how "no GPS signal" looks like for quite some time (might also be power saving).

[image]

abthediablo, 2021-05-26:
Well it seems that the bug was fixed with version 3.18.2 which I installed today.

And it seems the problem was to record a long track cause with 3.18.1 it was not a problem to start and continiue recording short tracks. But during a long cycling trip at the weekend the recording sopped after ~54km or 2:19h

Well and with 3.18.2 I was able to contunue this particular track or another track of ~288km and 13:30h.

mega-stoffel, 2021-05-27:
quick status update:
I can't get to the logs, the described method is way out of my capabilities.
I've installed the newer versions 3.18.1 & 3.18.2 via f-droid and haven't seen this behaviour anymore. But I didn't do excessive testing, just used the app on my daily trips. I will keep an eye on it.
Thank you! :-)

dennisguse, 2021-05-27:
@abthediablo @mega-stoffel thanks for testing it!
Glad that the issue is gone :partying_face: 

Just close this for now; if it appears again, please let us know!

closed 2021-06-03 by mega-stoffel