OpenTracksApp/OpenTracks #783: Add support for BLE continuous blood glucose monitors

gordonel, 2021-05-18:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I'd like to see CGM's supported in OpenTracks on the same level as other BLE sensors

**Describe alternatives you've considered**
Proprietary manufacturer-written apps (no-go for me)

**Additional context**
CGMs are entering the world of high-intensity training monitors. Companies like [Supersapiens]( https://www.supersapiens.com/en-GB/cart/) are offering CGMs designed with HIT in mind at a relatively affordable price. If this feature needs a sponsor, I'll consider it

dennisguse, 2021-05-18:
Adding a sensor is a a lot of (straight) forward work (only getting data from the sensor, no fancy calibration or such things).
Only requirement: the sensor provides a BLE profile.
See https://www.bluetooth.com/specifications/specs/

Actual task: 
* [ ] get info how to parse the data and the service UUID
* [ ] add a column to the TrackPoint table to store the data
* [ ] add a preference to select a sensor
* [ ] add internal sensor data passing (SensorDataSet)
* [ ] import/export: figure out how to handle GPX/KML
* [ ] implement UI: StatisticsRecordingFragment
* [ ] implement UI: chartView
* [ ] (optional) implement UI: chartView

And I need a sensor (physically) - the implementation of the power meter was done without and it took 2h hours to figure that it was almost correct ;)

gordonel, 2021-05-18:
What tools do you usually use to figure out if the sensor provides a BLE profile and get raw data?

dennisguse, 2021-05-19:
Good starting point: 
* BLExplorer (crashes on newer Android) https://f-droid.org/en/packages/org.ligi.blexplorer/
* bluetoothctl (linux command line)

Alternative while developing: debugging within OpenTracks
You could just change the service UUIDs of one of the working sensors ;)
(if you know what your sensor is doing)

dennisguse, 2021-06-22:
@gordonel I should have some time in July.
Can you get me such a sensor?

gordonel, 2021-06-22:
@dennisguse I think this can be de-prioritised for now. CGMs have been banned by UCI, and there seems to be little demand for it amongst non-cyclists

dennisguse, 2021-06-22:
Works for me; would have been a nice exercise ;)

Info: https://road.cc/content/tech-news/uci-bans-blood-glucose-monitoring-devices-races-283891

gordonel, 2021-06-22:
I'm thinking of buying a Supersapiens trial pack, but haven't decided yet